openbmc/openbmc #3944: build phosphor-power error with my system

YuanYuLin, 2023-08-03:
Hi all, 

I'm newbie in openbmc and I'm trying to add phosphor-power to my system.
following are my steps:
1. this is my recipes-phosphor folder structure.
recipes-phosphor/
├── packagegroups
│   └── packagegroup-obmc-apps.bbappend
├── power
│   ├── phosphor-power
│   │   ├── obmc
│   │   │   └── power-supply-monitor
│   │   │       ├── power-supply-monitor-0.conf
│   │   │       └── power-supply-monitor-1.conf
│   │   └── psu.json
│   └── phosphor-power_%.bbappend
└── state
    ├── phosphor-state-manager
    │   └── phosphor-service-monitor-default.json
    └── phosphor-state-manager_%.bbappend

2. content of phosphor-power_%.bbappend
FILESEXTRAPATHS:prepend := "${THISDIR}/${PN}:"

inherit obmc-phosphor-systemd

SRC_URI += "file://psu.json"

PACKAGECONFIG:append = " monitor-ng"

do_install:append(){
    install -D ${WORKDIR}/psu.json ${D}${datadir}/phosphor-power/psu.json
}
FILES:${PN} += "${datadir}/phosphor-power/psu.json"

PSU_MONITOR_ENV_FMT = "obmc/power-supply-monitor/power-supply-monitor-{0}.conf"
SYSTEMD_ENVIRONMENT_FILE:${PN}-monitor:append = " ${@compose_list(d, 'PSU_MONITOR_ENV_FMT', 'OBMC_POWER_SUPPLY_INSTANCES')}"
3. content of packagegroup-obmc-apps.bbappend
POWER_SERVICE_PACKAGES_RASPBERRYPI = " \
                                    phosphor-power-monitor \
                                    phosphor-power-systemd-links-monitor \
                                    phosphor-power-utils \
                                    phosphor-power \
                                  "

RDEPENDS:${PN}-extras:append = " \
                                          ${POWER_SERVICE_PACKAGES_RASPBERRYPI} \
                                          webui-vue \
                                         "
4. after build, I got this error
| [60/77] arm-openbmc-linux-gnueabi-g++ -marm -mfpu=vfp -mfloat-abi=hard -mcpu=arm1176jzf-s -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/van
ish/development/Adam/openbmc-build/raspberrypi/tmp/work/arm1176jzfshf-vfp-openbmc-linux-gnueabi/phosphor-power/1.0+git999-r1/recipe-sysroot -Iphosphor-power-supply/phosphor-psu-monitor.p -Iphosphor-power-supply 
-I../../../../../../workspace/sources/phosphor-power/phosphor-power-supply -I. -I../../../../../../workspace/sources/phosphor-power -flto=auto -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pc
h -Wextra -Wpedantic -Werror -std=c++20 -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/vanish/development/Adam/openbmc-build/raspberrypi/workspace/sources/phosphor-power=/usr/src/debug/phos
phor-power/1.0+git999-r1 -fdebug-prefix-map=/home/vanish/development/Adam/openbmc-build/raspberrypi/workspace/sources/phosphor-power=/usr/src/debug/phosphor-power/1.0+git999-r1 -fmacro-prefix-map=/home/vanish/de
velopment/Adam/openbmc-build/raspberrypi/tmp/work/arm1176jzfshf-vfp-openbmc-linux-gnueabi/phosphor-power/1.0+git999-r1/phosphor-power-1.0+git999=/usr/src/debug/phosphor-power/1.0+git999-r1 -fdebug-prefix-map=/ho
me/vanish/development/Adam/openbmc-build/raspberrypi/tmp/work/arm1176jzfshf-vfp-openbmc-linux-gnueabi/phosphor-power/1.0+git999-r1/phosphor-power-1.0+git999=/usr/src/debug/phosphor-power/1.0+git999-r1 -fdebug-pr
efix-map=/home/vanish/development/Adam/openbmc-build/raspberrypi/tmp/work/arm1176jzfshf-vfp-openbmc-linux-gnueabi/phosphor-power/1.0+git999-r1/recipe-sysroot= -fmacro-prefix-map=/home/vanish/development/Adam/ope
nbmc-build/raspberrypi/tmp/work/arm1176jzfshf-vfp-openbmc-linux-gnueabi/phosphor-power/1.0+git999-r1/recipe-sysroot= -fdebug-prefix-map=/home/vanish/development/Adam/openbmc-build/raspberrypi/tmp/work/arm1176jzf
shf-vfp-openbmc-linux-gnueabi/phosphor-power/1.0+git999-r1/recipe-sysroot-native= -fvisibility-inlines-hidden -DBOOST_ASIO_DISABLE_THREADS -DBOOST_ALL_NO_LIB -DBOOST_SYSTEM_NO_DEPRECATED -DBOOST_ERROR_CODE_HEADE
R_ONLY -DBOOST_COROUTINES_NO_DEPRECATION_WARNING -MD -MQ phosphor-power-supply/phosphor-psu-monitor.p/main.cpp.o -MF phosphor-power-supply/phosphor-psu-monitor.p/main.cpp.o.d -o phosphor-power-supply/phosphor-ps
u-monitor.p/main.cpp.o -c ../../../../../../workspace/sources/phosphor-power/phosphor-power-supply/main.cpp                                                                                                        
| FAILED: phosphor-power-supply/phosphor-psu-monitor.p/main.cpp.o                                                                                                                                                  
| arm-openbmc-linux-gnueabi-g++ -marm -mfpu=vfp -mfloat-abi=hard -mcpu=arm1176jzf-s -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/vanish/deve
lopment/Adam/openbmc-build/raspberrypi/tmp/work/arm1176jzfshf-vfp-openbmc-linux-gnueabi/phosphor-power/1.0+git999-r1/recipe-sysroot -Iphosphor-power-supply/phosphor-psu-monitor.p -Iphosphor-power-supply -I../../
../../../../workspace/sources/phosphor-power/phosphor-power-supply -I. -I../../../../../../workspace/sources/phosphor-power -flto=auto -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextr
a -Wpedantic -Werror -std=c++20 -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/vanish/development/Adam/openbmc-build/raspberrypi/workspace/sources/phosphor-power=/usr/src/debug/phosphor-pow
er/1.0+git999-r1 -fdebug-prefix-map=/home/vanish/development/Adam/openbmc-build/raspberrypi/workspace/sources/phosphor-power=/usr/src/debug/phosphor-power/1.0+git999-r1 -fmacro-prefix-map=/home/vanish/developmen
t/Adam/openbmc-build/raspberrypi/tmp/work/arm1176jzfshf-vfp-openbmc-linux-gnueabi/phosphor-power/1.0+git999-r1/phosphor-power-1.0+git999=/usr/src/debug/phosphor-power/1.0+git999-r1 -fdebug-prefix-map=/home/vanis
h/development/Adam/openbmc-build/raspberrypi/tmp/work/arm1176jzfshf-vfp-openbmc-linux-gnueabi/phosphor-power/1.0+git999-r1/phosphor-power-1.0+git999=/usr/src/debug/phosphor-power/1.0+git999-r1 -fdebug-prefix-map
=/home/vanish/development/Adam/openbmc-build/raspberrypi/tmp/work/arm1176jzfshf-vfp-openbmc-linux-gnueabi/phosphor-power/1.0+git999-r1/recipe-sysroot= -fmacro-prefix-map=/home/vanish/development/Adam/openbmc-bui
ld/raspberrypi/tmp/work/arm1176jzfshf-vfp-openbmc-linux-gnueabi/phosphor-power/1.0+git999-r1/recipe-sysroot= -fdebug-prefix-map=/home/vanish/development/Adam/openbmc-build/raspberrypi/tmp/work/arm1176jzfshf-vfp-
openbmc-linux-gnueabi/phosphor-power/1.0+git999-r1/recipe-sysroot-native= -fvisibility-inlines-hidden -DBOOST_ASIO_DISABLE_THREADS -DBOOST_ALL_NO_LIB -DBOOST_SYSTEM_NO_DEPRECATED -DBOOST_ERROR_CODE_HEADER_ONLY -
DBOOST_COROUTINES_NO_DEPRECATION_WARNING -MD -MQ phosphor-power-supply/phosphor-psu-monitor.p/main.cpp.o -MF phosphor-power-supply/phosphor-psu-monitor.p/main.cpp.o.d -o phosphor-power-supply/phosphor-psu-monito
r.p/main.cpp.o -c ../../../../../../workspace/sources/phosphor-power/phosphor-power-supply/main.cpp                                                                                                                
| In file included from ../../../../../../workspace/sources/phosphor-power/phosphor-power-supply/power_supply.hpp:3,                                                                                               
|                  from ../../../../../../workspace/sources/phosphor-power/phosphor-power-supply/psu_manager.hpp:3,
|                  from ../../../../../../workspace/sources/phosphor-power/phosphor-power-supply/main.cpp:16:
| ../../../../../../workspace/sources/phosphor-power/phosphor-power-supply/average.hpp:2:10: fatal error: org/open_power/Sensor/Aggregation/History/Average/server.hpp: No such file or directory
|     2 | #include <org/open_power/Sensor/Aggregation/History/Average/server.hpp>            
|       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| compilation terminated.

I also look into my recipe-sysroot of  phosphor-power build folder. 
There is no org/open_power in the include folder, how do I add it and what is this feature for?

Thanks

williamspatrick, 2023-08-03:
Something is misconfigured if you’re ending up needing org/open_power interfaces on a non-Power platform. It could be a problem with the underlying repository. 

@spinler @smccarney - Could you take a look?

williamspatrick, 2023-08-03:
We added those interfaces into the common repo but they are disabled by default, except on openpower systems. The repository should only use those interfaces when a feature requires them (and behind a meson option).

YuanYuLin, 2023-08-03:
@spinler @williamspatrick 
Thanks for your reply.
There are few questions that I want to know and give me some points that I can trace by myself, 
1. how do I add dependency? 
I had try to add this in my "phosphor-power_%.bbappend", but it still build failed with same error.
FILESEXTRAPATHS:prepend := "${THISDIR}/${PN}:"

inherit obmc-phosphor-systemd

DEPENDS += " \
  phosphor-dbus-interfaces \
  sdbusplus \
  systemd \
  "

SRC_URI += "file://psu.json"

PACKAGECONFIG:append = " monitor"

do_install:append(){
    install -D ${WORKDIR}/psu.json ${D}${datadir}/phosphor-power/psu.json
}
FILES:${PN} += "${datadir}/phosphor-power/psu.json"

PSU_MONITOR_ENV_FMT = "obmc/power-supply-monitor/power-supply-monitor-{0}.conf"
SYSTEMD_ENVIRONMENT_FILE:${PN}-monitor:append = " ${@compose_list(d, 'PSU_MONITOR_ENV_FMT', 'OBMC_POWER_SUPPLY_INSTANCES')}"

2. I also checked phosphor-dbus-interfaces, the gen did not ran with my system,
    from README.md, and there is a way to gen by manually.
    how do I run it from recipe(bbappends)?

Thanks

spinler, 2023-08-03:
For now, looks like you need to add something like:
```
OBMC_ORG_YAML_SUBDIRS += " org/open_power"
```
somewhere in your tree, for example like here: https://github.com/openbmc/openbmc/blob/master/meta-openpower/conf/distro/include/openpower-distrovars.inc#L6

I can working on fixing it so that won't be necessary in the future.

williamspatrick, 2023-08-03:
```
PACKAGECONFIG:append = " monitor"
```

This currently only works with open-power machines.

It looks like "monitor-ng" does not use any `org.open_power` d-bus interfaces.  You might want to look at that instead.

YuanYuLin, 2023-08-05:
It works, Thanks.

closed 2023-08-05 by YuanYuLin

spinler, 2023-08-10:
It's likely we're going to be able to get away with not having to supply the trailing power supply 30 second averages and maximums anymore, in which case we'll remove this org.open_power interface from the code.  If that turns out not to be the case, we'll add a meson option for both monitor and monitor-ng to not use it.