openbmc/openbmc #3946: Watchdog timeout reset error after update

vsytch, 2023-08-13:
**Unexpected behavior you saw**
During flash update in https://github.com/openbmc/openbmc/blob/master/meta-phosphor/recipes-phosphor/initrdscripts/files/obmc-shutdown.sh#L68 I observe the following error:
```
    Resetting watchdog timeouts to -T 15
    watchdog: invalid number ' 15
```
**Expected behavior**
No error message should be printed.

**To Reproduce**:
Flash new image.

**Additional context**
A similar problem was fixed with https://github.com/openbmc/openbmc/commit/321a95f8e0413365a569da6438ac748591bd1a68. I applied the same type of fix:
```
--- a/meta-phosphor/recipes-phosphor/initrdscripts/files/obmc-shutdown.sh
+++ b/meta-phosphor/recipes-phosphor/initrdscripts/files/obmc-shutdown.sh
@@ -65,7 +65,8 @@ then
                        if test -n "$wdrst"
                        then
                                echo "Resetting watchdog timeouts to $wdrst"
-                               watchdog "$wdrst" -F /dev/watchdog &
+                               # shellcheck disable=SC2086
+                               watchdog $wdrst -F /dev/watchdog &
                                sleep 1
                                # Kill the watchdog daemon, setting a timeout
                                # for the remaining shutdown work
                                
```
but I hit a new error, saying `/dev/watchdog` is busy (sorry, lost my last update log). I'm not clear if that is expected at this stage.

mdmillerii, 2023-08-13:
Good catch this is definitely needed.   Can you open a gerrit?   Are you under a cla?

The expectation is the kill on https://github.com/openbmc/openbmc/blob/321a95f8e0413365a569da6438ac748591bd1a68/meta-phosphor/recipes-phosphor/initrdscripts/files/obmc-shutdown.sh#L64 should kill the previous watchdog daemon that had the watchdog open.   The kill a few lines later is also needed to kill this after the timeout is reset.

williamspatrick, 2023-08-14:
remote:   https://gerrit.openbmc.org/c/openbmc/openbmc/+/65904 meta-phosphor: initrdscripts: fix watchdog reset [NEW]

mdmillerii, 2023-08-14:
Note if you tried to invoke the shutdown script manually instead of systemd termination I'd expect the watchdog to be busy by systemd otherwise it requires more investigation

closed 2023-08-14 by geissonator