openbmc/openbmc #3947: BMC stopped updating the flash image after commit 2a25492

gluhow, 2023-08-16:
After commit   https://github.com/openbmc/openbmc/commit/2a25492c13e2b768f94b864a51f84e82e4238aef my bmc stopped updating the flash image.

To reproduce the error:
1) Update your bmc to this commit or newer
2) Try to update the BMC image via the standard web Update firmware.
BMC reboots, but flash does not update. In qemu the problem can also be observed

My update log:
```
[  OK  ] Reached target System Reboot.
[  356.400858] watchdog: watchdog0: watchdog did not stop!
[  356.514184] systemd-shutdown[1]: Using hardware watchdog 'aspeed_wdt', version 0, device /dev/watchdog0
[  356.524082] systemd-shutdown[1]: Watchdog running with a timeout of 1min.
[  356.599543] systemd-shutdown[1]: Syncing filesystems and block devices.
[  356.615690] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[  356.682298] systemd-journald[125]: Received SIGTERM from PID 1 (systemd-shutdow).
[  356.734316] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[  356.810308] systemd-shutdown[1]: Unmounting file systems.
[  356.831011] (sd-remount)[489]: Remounting '/' read-only with options 'lowerdir=run/initramfs/ro,upperdir=run/initramfs/rw/cow,workdir=run/initramfs/rw/work'.
[  356.870153] (sd-remount)[490]: Remounting '/run/initramfs/rw' read-only with options ''.
[  356.881372] systemd-shutdown[1]: All filesystems unmounted.
[  356.887157] systemd-shutdown[1]: Deactivating swaps.
[  356.892938] systemd-shutdown[1]: All swaps deactivated.
[  356.898373] systemd-shutdown[1]: Detaching loop devices.
[  356.933118] systemd-shutdown[1]: All loop devices detached.
[  356.938909] systemd-shutdown[1]: Stopping MD devices.
[  356.947259] systemd-shutdown[1]: All MD devices stopped.
[  356.953029] systemd-shutdown[1]: Detaching DM devices.
[  356.961341] systemd-shutdown[1]: All DM devices detached.
[  356.966951] systemd-shutdown[1]: All filesystems, swaps, loop devices, MD devices and DM devices detached.
[  357.012908] systemd-shutdown[1]: Failed to move /run/initramfs to /: Invalid argument
[  357.021293] systemd-shutdown[1]: Failed to switch root to "/run/initramfs": Invalid argument
[  357.031422] systemd-shutdown[1]: Syncing filesystems and block devices.
[  357.045755] systemd-shutdown[1]: Rebooting.
[  357.051694] reboot: Restarting system
```

matnick, 2023-08-16:
Same problem. Same logs.

mdmillerii, 2023-08-16:
By design when using our static mtd layout, our initrd [scripts](https://github.com/openbmc/openbmc/blob/e99fc5b5e41d1f0aa3d9dc5fa37a18306c720f24/meta-phosphor/recipes-phosphor/initrdscripts/files) [init](https://github.com/openbmc/openbmc/blob/e99fc5b5e41d1f0aa3d9dc5fa37a18306c720f24/meta-phosphor/recipes-phosphor/initrdscripts/files/obmc-init.sh#L22) script prepares `/run/initramfs/shutdown` environment according to the [systemd initrd interface](https://systemd.io/INITRD_INTERFACE/) to allow the overlay to be [unmounted](https://github.com/openbmc/openbmc/blob/e99fc5b5e41d1f0aa3d9dc5fa37a18306c720f24/meta-phosphor/recipes-phosphor/initrdscripts/files/obmc-shutdown.sh#L23..L31), freeing the underlying mtd storage and then [calling]https://github.com/openbmc/openbmc/blob/e99fc5b5e41d1f0aa3d9dc5fa37a18306c720f24/meta-phosphor/recipes-phosphor/initrdscripts/files/obmc-shutdown.sh#L52) the update script to write the new images into the flash (mtd) storage.

In the above yocto update systemd was bumped from 253.3 to 254.0
```
SRCREV = "ed18c2ab79e8b94182d5dcf31d58457763f3e3e1"
SRCBRANCH = "v253-stable"
SRCREV = "994c7978608a0bd9b317f4f74ff266dd50a3e74e"
SRCBRANCH = "v254-stable"
```
Since then [v254.1 was tagged](https://github.com/systemd/systemd-stable/compare/v254...v254-stable) and I see it has further [changes](https://github.com/systemd/systemd-stable/commit/6068122836726aadbb0a3ebae2290723774b8071) but I don't know they will fix anything and may expose further [breakage](https://github.com/systemd/systemd-stable/commit/60046bf2a9f750a3ef17aab5948b9e5ba8734265) [aka](https://github.com/systemd/systemd-stable/commit/d709d1b20e2e15ee2ae1b44de94d493e17834235) , as "/run" is not accessible from "/run/initramfs" bind mount chroot without mounting "/run" underneath it.

LeeTroy, 2023-08-30:
> By design when using our static mtd layout, our initrd [scripts](https://github.com/openbmc/openbmc/blob/e99fc5b5e41d1f0aa3d9dc5fa37a18306c720f24/meta-phosphor/recipes-phosphor/initrdscripts/files) [...]

This submitted resolves the failed to switch root to /run/initramfs issue, but firmware upgrade is failed due to mtd is still mounted.
```
[ 1397.787445] systemd-shutdown[1]: Unmounting file systems.
[ 1397.802865] (sd-remount)[502]: Remounting '/' read-only with options 'lowerdir=run/initramfs/ro,upperdir=run/initramfs/rw/cow,wor
kdir=run/initramfs/rw/work'.
[ 1397.827174] (sd-remount)[503]: Remounting '/run/initramfs/rw' read-only with options ''.
[ 1397.837708] systemd-shutdown[1]: All filesystems unmounted.
[ 1397.843982] systemd-shutdown[1]: Deactivating swaps.
[ 1397.849739] systemd-shutdown[1]: All swaps deactivated.
[ 1397.855597] systemd-shutdown[1]: Detaching loop devices.
[ 1397.867106] systemd-shutdown[1]: All loop devices detached.
[ 1397.873494] systemd-shutdown[1]: Stopping MD devices.
[ 1397.880149] systemd-shutdown[1]: All MD devices stopped.
[ 1397.886115] systemd-shutdown[1]: Detaching DM devices.
[ 1397.892519] systemd-shutdown[1]: All DM devices detached.
[ 1397.898780] systemd-shutdown[1]: All filesystems, swaps, loop devices, MD devices and DM devices detached.
[ 1397.911708] systemd-shutdown[1]: Successfully changed into root pivot.
[ 1397.919174] systemd-shutdown[1]: Returning to initrd...
shutdown: reboot --log-level=6 --timeout=90000000us --log-target=kmsg --log-color --exit-code=0
+ awk '/oldroot|mnt/ { print $2 }'
+ sort -r
+ IFS= read -r f
// SKIP ...
update: --clean-saved-files
/dev/mtdblock4 /ro squashfs ro,relatime,errors=continue 0 0
ERROR: Device mtd4 [ 1398.258271] watchdog: watchdog0: watchdog did not stop!
is mounted, image-rofs is busy.        <-------
Flash update failed to flash these images:    <-------
/run/initramfs/image-kernel
/run/initramfs/image-rofs
Resetting watchdog timeouts to -T 15
watchdog: can't open '/dev/watchdog': Device or resource busy
Remaining mounts:
tmpfs / tmpfs rw,nosuid,nodev,mode=755 0 0
/dev/mtdblock4 /ro squashfs ro,relatime,errors=continue 0 0
/dev/mtdblock5 /rw jffs2 ro,relatime 0 0
dev /dev devtmpfs rw,relatime,size=430220k,nr_inodes=107555,mode=755 0 0
tmpfs /dev/shm tmpfs rw,nosuid,nodev 0 0
devpts /dev/pts devpts rw,relatime,gid=5,mode=620,ptmxmode=666 0 0
sys /sys sysfs rw,relatime 0 0
cgroup2 /sys/fs/cgroup cgroup2 rw,nosuid,nodev,noexec,relatime,nsdelegate,memory_recursiveprot 0 0
bpf /sys/fs/bpf bpf rw,nosuid,nodev,noexec,relatime,mode=700 0 0
debugfs /sys/kernel/debug debugfs rw,nosuid,nodev,noexec,relatime 0 0
```

mdmillerii, 2023-08-30:
I have been experimenting with several possible changes and now understand what's happening but need to summarize the options.    Upstream had identified some issues that have been addressed in v254.1 stable backports (and added testcases) but some additional changes will be required in our side.   

The unmount is failing in v254 because there are additional mounts confusingly on top that's preventing the unmount.  The v254.1 tag avoids this over mount.

The easiest way to  a working build today is to revert to the prior (v253.3) systemd recipe with something like 

```shell
git checkout --no-overlay 2a25492^ -- poky/meta/recipes-core/systemd
```

mdmillerii, 2023-08-30:
I have pushed a branch [initramfs-workaround](https://github.com/mdmillerii/openbmc/compare/master...mdmillerii:openbmc:initramfs-workaround?expand=0) to GitHub that "shows my work" albeit rebased to show the end results.

The `mount under /run/mnt` option work but systemd unmounts it out of order, so we could miss that / is still mounted and the filesystems are busy.  The simplest is probably to just unmount /ro and /rw, but doing the bind mount if we mounted a tmpfs is also a good candidate for submission to gerrit.

iwoloschin, 2023-09-26:
Mentioned in [Discord](https://discord.com/channels/775381525260664832/867820390406422538/1156161608071004251):

https://github.com/Nuvoton-Israel/openbmc/commit/587525e8e038228c0622d4c86e1842f9d62e7872

mdmillerii, 2023-10-12:
67046: initfs: Adapt shutdown unmount to systemd v254 | https://gerrit.openbmc.org/c/openbmc/openbmc/+/67046