openbmc/openbmc #3949: Update image using Redfish curl - TFTP

Saravanans2001, 2023-08-28:
Hi all,
   To upgrade the firmware, I am using Redfish and the following curl command over TFTP.

curl -b cjar -k -u root:0penBmc -H "Content-Type: application/json" -X POST -d '{"TransferProtocol":"TFTP","ImageURI":"192.168.1.80/bmc-phosphor-image-evb-ast2600-20230824064748.static.mtd.tar"}' [172.20.48.110/redfish/v1/UpdateService/Actions/UpdateService.SimpleUpdate](https://172.20.48.110/redfish/v1/UpdateService/Actions/UpdateService.SimpleUpdate)

In postman tool ,I initially encounter the 503 service unavailable error before encountering the 500 internal server error.
[image]

1.How to overcome the internal service error for tftp?
2.Kindly help me,If i am done wrong .

williamspatrick, 2023-09-25:
Have you tried adding `tftp://` to the ImageURI?

Saravanans2001, 2023-10-19:
@williamspatrick
Yes It is working

closed 2023-10-19 by williamspatrick