openbmc/openbmc #3951: linux 6.5 fails do_image_cpio step

shenki, 2023-08-31:
We are attempting to move to a newer kernel version. In 6.5, the arm32 maintainers moved the device trees to a set of vendor specific subdirectories, meaning a device tree that was found at `arch/arm/boot/dts/aspeed-ast2600-evb.dts` is now at `arch/arm/boot/dts/aspeed/aspeed-ast2600-evb.dts`. 

Upstream openembedded-core seems to solve this by adding the directory to the KERNEL_DEVICETREE variable. This works for most of openbmc's machines, but fails for two Facebook machines: yosemite4 and bletchley.

The output of the build failure is below:

```
04:26:36 | DEBUG: Python function set_image_size finished
04:26:36 | DEBUG: Executing shell function do_image_cpio
04:26:36 | 214160 blocks
04:26:36 | 2 blocks
04:26:36 | NOTE: Did not find initramfs image: /data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/yosemite4/build/deploy/images/yosemite4/obmc-phosphor-image-yosemite4-20230830042151.cpio.zst-yosemite4.cpio.lz4
04:26:36 | NOTE: Did not find initramfs image: /data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/yosemite4/build/deploy/images/yosemite4/obmc-phosphor-image-yosemite4-20230830042151.cpio.zst-yosemite4.cpio.lzo
04:26:36 | NOTE: Did not find initramfs image: /data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/yosemite4/build/deploy/images/yosemite4/obmc-phosphor-image-yosemite4-20230830042151.cpio.zst-yosemite4.cpio.lzma
04:26:36 | NOTE: Did not find initramfs image: /data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/yosemite4/build/deploy/images/yosemite4/obmc-phosphor-image-yosemite4-20230830042151.cpio.zst-yosemite4.cpio.xz
04:26:36 | NOTE: Did not find initramfs image: /data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/yosemite4/build/deploy/images/yosemite4/obmc-phosphor-image-yosemite4-20230830042151.cpio.zst-yosemite4.cpio.zst
04:26:36 | NOTE: Did not find initramfs image: /data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/yosemite4/build/deploy/images/yosemite4/obmc-phosphor-image-yosemite4-20230830042151.cpio.zst-yosemite4.cpio.gz
04:26:36 | NOTE: Did not find initramfs image: /data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/yosemite4/build/deploy/images/yosemite4/obmc-phosphor-image-yosemite4-20230830042151.cpio.zst-yosemite4.ext2.gz
04:26:36 | NOTE: Did not find initramfs image: /data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/yosemite4/build/deploy/images/yosemite4/obmc-phosphor-image-yosemite4-20230830042151.cpio.zst-yosemite4.cpio
04:26:36 | NOTE: Found initramfs image: obmc-phosphor-image-yosemite4-20230830042151.cpio.zst
04:26:36 | FATAL ERROR: Couldn't open "/data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/yosemite4/build/deploy/images/yosemite4/aspeed/aspeed-bmc-facebook-yosemite4.dtb": No such file or directory
04:26:36 | uboot-mkimage: Can't open /data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/yosemite4/build/deploy/images/yosemite4/obmc-phosphor-image-yosemite4-20230830042151.cpio.zst.fitImage.tmp: No such file or directory
04:26:36 | Error: Bad parameters for FIT image type
04:26:36 | Usage: uboot-mkimage [-T type] -l image
04:26:36 |           -l ==> list image header information
04:26:36 |           -T ==> parse image file as 'type'
04:26:36 |           -q ==> quiet
04:26:36 |        uboot-mkimage [-x] -A arch -O os -T type -C comp -a addr -e ep -n name -d data_file[:data_file...] image
04:26:36 |           -A ==> set architecture to 'arch'
04:26:36 |           -O ==> set operating system to 'os'
04:26:36 |           -T ==> set image type to 'type'
04:26:36 |           -C ==> set compression type 'comp'
04:26:36 |           -a ==> set load address to 'addr' (hex)
04:26:36 |           -e ==> set entry point to 'ep' (hex)
04:26:36 |           -n ==> set image name to 'name'
04:26:36 |           -R ==> set second image name to 'name'
04:26:36 |           -d ==> use image data from 'datafile'
04:26:36 |           -x ==> set XIP (execute in place)
04:26:36 |           -s ==> create an image with no data
04:26:36 |           -v ==> verbose
04:26:36 |        uboot-mkimage [-D dtc_options] [-f fit-image.its|-f auto|-f auto-conf|-F] [-b <dtb> [-b <dtb>]] [-E] [-B size] [-i <ramdisk.cpio.gz>] fit-image
04:26:36 |            <dtb> file is used with -f auto, it may occur multiple times.
04:26:36 |           -D => set all options for device tree compiler
04:26:36 |           -f => input filename for FIT source
04:26:36 |           -i => input filename for ramdisk file
04:26:36 |           -E => place data outside of the FIT structure
04:26:36 |           -B => align size in hex for FIT structure and header
04:26:36 |           -b => append the device tree binary to the FIT
04:26:36 |           -t => update the timestamp in the FIT
04:26:36 | Signing / verified boot options: [-k keydir] [-K dtb] [ -c <comment>] [-p addr] [-r] [-N engine]
04:26:36 |           -k => set directory containing private keys
04:26:36 |           -K => write public keys to this .dtb file
04:26:36 |           -g => set key name hint
04:26:36 |           -G => use this signing key (in lieu of -k)
04:26:36 |           -c => add comment in signature node
04:26:36 |           -F => re-sign existing FIT image
04:26:36 |           -p => place external data at a static position
04:26:36 |           -r => mark keys used as 'required' in dtb
04:26:36 |           -N => openssl engine to use for signing
04:26:36 |           -o => algorithm to use for signing
04:26:36 |        uboot-mkimage -V ==> print version information and exit
04:26:36 | Use '-T list' to see a list of available image types
04:26:36 | Long options are available; read the man page for details
04:26:36 | WARNING: exit code 1 from a shell command.
04:26:36 NOTE: recipe obmc-phosphor-image-1.0-r0: task do_image_cpio: Failed
04:26:36 ERROR: Task (/data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/yosemite4/openbmc/meta-phosphor/recipes-phosphor/images/obmc-phosphor-image.bb:do_image_cpio) failed with exit code '1'```

shenki, 2023-08-31:
A WIP change that can be used to debug this issue is found here https://gerrit.openbmc.org/c/openbmc/openbmc/+/65884

shenki, 2023-08-31:
@williamspatrick fyi

williamspatrick, 2023-08-31:
There is a fitimage.bbclass that I haven’t had a chance to upstream and that likely needs similar changes to what was done in kernel-fitimage.bbclass.

geissonator, 2023-09-08:
I missed that https://gerrit.openbmc.org/c/openbmc/openbmc/+/65884/10/poky/meta/classes-recipe/kernel-fitimage.bbclass had a workaround for this issue. I added a similar one to https://gerrit.openbmc.org/c/openbmc/openbmc/+/65884/10/meta-phosphor/classes/fit-image.bbclass and it all at least builds now.

geissonator, 2023-09-11:
> How should we proceed?

With your change to that poky file + my change to our openbmc file, https://gerrit.openbmc.org/c/openbmc/openbmc/+/65884 made it through build and HW CI. I'll look into sending your change upstream to poky. Once we get agreement there on what the fix should look like, we can apply it to our file too.

shenki, 2023-09-11:
A good plan, thanks @geissonator

geissonator, 2023-09-11:
ok, well I further confused myself a bit on this one. The changes put in https://gerrit.openbmc.org/c/openbmc/openbmc/+/65884/10/poky/meta/classes-recipe/kernel-fitimage.bbclass were actually not needed to fix the non-facebook systems like I thought they were. So really all we need is the one line fix in https://gerrit.openbmc.org/c/openbmc/openbmc/+/65884/11/meta-phosphor/classes/fit-image.bbclass. 

I tried to identify which commits in upstream poky fixed it for the non-facebook systems and although I found a few that looked relevant, when I tried to manually apply them to our version of fit-image.bbclass, they didn't seem to help. So I'll let @williamspatrick take a look at the one-liner and verify all is still good with the image on his systems and I think we can just move forward with putting 6.5 into openbmc master (will need to move 6.15 from Joels fork into openbmc/linux) . I removed the un-needed change from the poky file in the gerrit commit.

williamspatrick, 2023-09-20:
Updated https://gerrit.openbmc.org/c/openbmc/openbmc/+/66610/3 with a more complete fix, which hopefully can be integrated into `kernel-fitimage.bbclass` upstream also.