openbmc/openbmc #3952: ObjectMapper Call failed: The resource is not found

renweihang, 2023-09-02:
1. There is a normal DBus , and the corresponding path and interface can be viewed using `busctl introspect` command, but why does "ObjectMapper GetObject " indicate failure?
2. Why is it that after restarting the ObjectMapper service, "ObjectMapper GetObjec" can successfully retrieve the DBus information?

    Thank you very much!  Here are the detailed information:
```
root@g50:~# busctl introspect xyz.openbmc_project.CPLDManager /xyz/openbmc_project/inventory/item/psu/psu1 xyz.openbmc_project.Inventory.Item
NAME                               TYPE      SIGNATURE RESULT/VALUE FLAGS
.Hotplug                           property  b         false        emits-change
.Present                           property  b         true         emits-change
root@g50:~# busctl call xyz.openbmc_project.ObjectMapper /xyz/openbmc_project/object_mapper xyz.openbmc_project.ObjectMapper GetObject sas /xyz/openbmc_project/inventory/item/psu/psu1 1 xyz.openbmc_project.Inventory.Item
Call failed: The resource is not found.
root@g50:~# 
root@g50:~# busctl |grep mapper -i
:1.12                                              312 mapperx         root             :1.12         xyz.openbmc_project.ObjectMapper.service                                  -       -
:1.34                                              375 mapper          root             :1.34         mapper-wait@-xyz-openbmc_project-led-groups-bmc_booted.service            -       -
:1.95                                              695 mapper          root             :1.95         mapper-wait@-xyz-openbmc_project-led-groups.service                       -       -
xyz.openbmc_project.ObjectMapper                   312 mapperx         root             :1.12         xyz.openbmc_project.ObjectMapper.service                                  -       -
root@g50:~# systemctl restart xyz.openbmc_project.ObjectMapper
root@g50:~# busctl call xyz.openbmc_project.ObjectMapper /xyz/openbmc_project/object_mapper xyz.openbmc_project.ObjectMapper GetObject sas /xyz/openbmc_project/inventory/item/psu/psu1 1 xyz.openbmc_project.Inventory.Item
a{sas} 1 "xyz.openbmc_project.CPLDManager" 4 "org.freedesktop.DBus.Introspectable" "org.freedesktop.DBus.Peer" "org.freedesktop.DBus.Properties" "xyz.openbmc_project.Inventory.Item"
root@g50:~#
```

mine260309, 2023-09-02:
> :1.34                                              375 mapper          root             :1.34         ***@***.***_project-led-groups-bmc_booted.service            -       -
> :1.95                                              695 mapper          root             :1.95         ***@***.***_project-led-groups.service                       -       -

>
> There is a normal DBus , and the corresponding path and interface can be viewed using busctl introspect command, but why does "ObjectMapper GetObject " indicate failure?
>
> Why is it that after restarting the ObjectMapper service, "ObjectMapper GetObjec" can successfully retrieve the DBus information?

Usually this indicates an issue in the service code that the signals
like interfacesAdded/Removed are not correctly emitted, so you get an
inconsistency between DBus and mapper.
Restarting mapper makes a full introspection of the DBus and you get
the consistency back.

williamspatrick, 2023-09-02:
This is a support question, not a bug. The question is about non-upstreamed code. Closing.

closed 2023-09-02 by williamspatrick