openbmc/openbmc #3956: /dev/ipmi0 interface not present in BMC board

vijaysai, 2023-09-12:
I am trying to compile openbmc source code for evb-2600 setup, once the phosphor Image is built, and flashed into BMC flash chip, the Image is booting fine and able to see linux shell, but there is no /dev/ipmi interfaces in BMC board, even blue screen which shows BMC management page, where it  updates information about BMC is all junk, if I set wait for BMC in bluescreen, in next there is a halt.even postcodes are not getting updated there is only 00 postcode.

williamspatrick, 2023-09-12:
`/dev/IPMI` would never be present on a BMC as that is a host-side device driver. The host could use `/dev/ipmi` to talk to its own BMC but the BMC side of the interface is different.

williamspatrick, 2023-09-12:
You mentioned an evb and then started referencing post codes. I’m not sure what system configuration you are attempting but an evb image is for the Aspeed evaluation board, which has no managed host. If you’re somehow trying to put that card into a system you’ll need to write code pertaining to your specific configuration.

closed 2023-09-12 by williamspatrick