openbmc/openbmc #3957: ipmitool sel list and fru list

Govindharajen, 2023-09-13:
Hi, I'm trying to including the ipmitool sel list and fru list in the same image. But it cant work at the same. When i include the fru changes, the fru list is shows and sel list is showing no entries and if the fru changes is removed, the sel entries are shows like undermined hardware failure but fru is not working it shows device not found. If it is possible to enable the both the list? Im having the error like below..
[image]

williamspatrick, 2023-09-25:
Please give details on how you are configuring this that only one works at a time.  These IPMI handlers are unrelated, so I don't know how enabling one would make the other not work.

closed 2023-09-25 by williamspatrick

Govindharajen, 2023-09-25:
Ok, Then why the fru list shows the device not found.. I generated the bin file for the fru list and wrote the bin file in eeprom also but it cant shows the exact information?

Govindharajen, 2023-09-25:
> Please give details on how you are configuring this that only one works at a time. These IPMI handlers are unrelated, so I don't know how enabling one would make the other not work.

Ok, Then why the fru list shows the device not found.. I generated the bin file for the fru list and wrote the bin file in eeprom also but it cant shows the exact information?

williamspatrick, 2023-09-25:
You originally wrote that you can’t get them both working. Is the problem that only SEL or FRU works at a time?  Or is the problem you can’t get FRU working?

I don’t know what “I generated the bin file for FRU list” means.  What does entity-manager parse for your eeproms?

The “device not found” errors seem to be related to DIMMs and cpus. Typically the BMC can’t get this information directly but relies on a BIOS component (such as smbios-mdr) to provide it. Did you enable a provider for this data?

Govindharajen, 2023-09-25:
> You originally wrote that you can’t get them both working. Is the problem that only SEL or FRU works at a time? Or is the problem you can’t get FRU working?
>
> I don’t know what “I generated the bin [...]

> You originally wrote that you can’t get them both working. Is the problem that only SEL or FRU works at a time? Or is the problem you can’t get FRU working?
>
> I don’t know what “I generated the bin [...]

Now I have added my fru data in eeprom (I2C -7 or 7-0050) using dd command.
command : dd if=/etc/fru/baseboard.fru.bin of=/sys/bus/i2c/devices/7-0050/eeprom

[image]

How can I configure ipmitool to show fru details from eeprom.

williamspatrick, 2023-09-25:
Once again:

> What does entity-manager parse for your eeproms?

Govindharajen, 2023-09-25:
> Once again:
> 
> > What does entity-manager parse for your eeproms?

I don't know what you are asking about entity-manager? Kindly help me to understand this problem..!!

williamspatrick, 2023-09-25:
There are a number of pieces here and code that needs machine configuration. You can’t expect to simply write an eeprom and “magic happens”.

If you have an eeprom in IPMI format, there is a process called fru-device that will parse it. You can look for this process in the busctl to inspect the dbus objects it created. 

After that, entity-manager has json that defines how to identify your eeprom as a particular FRU (or “entity”) and turn the eeprom records into various inventory objects. Again, busctl can be used to see what entity-manager decided. If you haven’t written an entity-manager json for your board, it likely isn’t creating inventory objects. 

After that, the IPMI code has a json or yaml file to define the map from inventory dbus records to FRU numbers from an IPMI perspective. You have to tell it which inventory object paths you want as which IPMI FRU record. You probably need to customize this for your board as well. 

This has gone way outside the scope of “there might be a bug” and into “I am looking for help/support”. Please transition to discord for further assistance.

williamspatrick, 2023-09-29:
Oh, good to know.  I don't know why those are related.  I'd like to see us deprecate the rsyslog path in the not too distant future.

iwoloschin, 2023-10-02:
Took a bit longer than I wanted as I wanted to understand the problem better, but I've filed an issue here: https://github.com/openbmc/phosphor-host-ipmid/issues/197