openbmc/openbmc #3959: Virtual Media Mount Installation Windows Report Driver Exception Error

Hong1230, 2023-09-26:
When using virtual media to mount a Windows Server 2022 image (5.1G), the mounting process is normal, but during the mounting boot installation, a driver loading failure will be reported. The error message is that the required media driver for the computer is missing.

May I ask if there is an issue with OpenBMC's support for Windows x86, or is this Windows image larger than 4G?

How should we solve it? Do we need to modify the kernel driver and how to do so?

Can I modify the window image to complete the mount boot installation? How to modify it?

texroemer, 2023-10-02:
Hi @Hong1230 

Try converting the ISO to IMG file, attach IMG file and start over the OS installation.

Thanks
Tex

mohammedjavitham, 2024-03-27:
https://lists.ozlabs.org/pipermail/openbmc/2020-October/023701.html