openbmc/openbmc #3961: bletchley: Cannot find any SPDX file for document http://spdx.org/spdxdoc/kernel-module-mdio-netlink-6.5.4

geissonator, 2023-10-26:
**Unexpected behavior you saw**
Something went into yocto recently that we picked up that is not allowing new kernels to build for the bletchley platform. This is preventing us from being able to pick up new kernel bumps in master. It seems we can build once with a fresh build and utilize sstate after that which is why things are ok until we try to pick up a new kernel.

https://gerrit.openbmc.org/c/openbmc/openbmc/+/67402 is a recent example of this issue.

**Expected behavior**
Code should build

**To Reproduce**
Steps to reproduce the behavior:
1. Cherry pick above commit and try to build bletchley

**Screenshots**
N/A

**OpenBMC Information:**
Latest master

**Additional context**
There was some discord discussion and it is theorized that bletchley is the only platform to utilize kernel modules.

amboar, 2023-10-26:
Based on a brief investigation I think the module's metadata isn't setup correctly and a stale sstate is deployed. I'm not yet sure what the fix is.

amboar, 2023-10-30:
Okay, I've reproduced the problem locally and it appears to be fixed by:

```diff
diff --git a/meta-openembedded/meta-networking/recipes-support/mdio-tools/mdio-tools_1.3.0.bb b/meta-openembedded/meta-networking/recipes-support/mdio-tools/mdio-tools_1.3.0.bb
index cd4df3da05cb..0c64889bc390 100644
--- a/meta-openembedded/meta-networking/recipes-support/mdio-tools/mdio-tools_1.3.0.bb
+++ b/meta-openembedded/meta-networking/recipes-support/mdio-tools/mdio-tools_1.3.0.bb
@@ -1,6 +1,6 @@
 require mdio-tools.inc
 
-DEPENDS += "libmnl"
+DEPENDS += "virtual/kernel libmnl"
 
 S = "${WORKDIR}/git"
 
```

amboar, 2023-10-30:
https://patchwork.yoctoproject.org/project/oe/patch/20231030034753.312182-1-andrew@codeconstruct.com.au/

amboar, 2023-10-31:
The patch above has been applied to master-next upstream:

https://git.openembedded.org/meta-openembedded/commit/?h=master-next&id=668cf43b21e27faa34b7c3c7133a480a9e4e480f

We've merged into OpenBMC as well to unblock kernel bumps:

https://github.com/openbmc/openbmc/commit/fd996948bbcdd32d1bd91b4fe4c9c873d04dc566

Our downstream patch should fall out in the next yocto update.

closed 2023-10-31 by amboar