openbmc/openbmc #3962: perl_5.38.0.bb:do_compile

zk-807, 2023-11-01:
在bitbake obmc-phosphor-image期间报错如下
 make[2]: Entering directory '/hERROR: Task (virtual:native:/home/zhangkai/桌面/openbmc/meta/recipes-devtools/perl/perl_5.38.0.bb:do_compile) failed with exit code '1'

mdmillerii, 2023-11-01:
I just completed a build of palmetto that compled within the last 8 hours (which includes perl 5.38.0) , so I doubt anything upstream is broken.   Check the build logs, look for out of memory kill (and retry, maybe adjusting the PARALLEL_MAKE and the bitbake parallel I don't immediately recall to reduce stress on your build machine.

amboar, 2023-11-01:
it's perl-native as well, so it may be something to do with the build environment (which we have no information about).

@zk-807 The bug template asks for many details and it appears you've just ignored and deleted it. It's next to impossible to help you with the (lack of) information you've provided. Here's what the template asks for:

```
**Note on Bugs**
Please note that questions or feature requests should be discussed on the
mailing list or Discord. Please see https://github.com/openbmc/openbmc#contact
for info on how to contact us. Github Issues in openbmc/openbmc are for bugs
only. Please remove this section when opening a bug.

**Unexpected behavior you saw**
A clear and concise description of what unexpectedly happened.

**Expected behavior**
A clear and concise description of what you expected to happen.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**OpenBMC Information:**
 - /etc/os-release:
 - uname -a:
 - Any additional patches or changes:

**Additional context**
Add any other context about the problem here.
```

Please re-open the issue once you've provided those details.

closed 2023-11-01 by amboar