openbmc/openbmc #3967: timedatectl takes too long to execute

jack206, 2023-11-20:
root@BMC:~# busctl |cat |grep "org.freedesktop.timedate1"
org.freedesktop.timedate1                            - -               -                (activatable) -                         -
root@BMC:~# time timedatectl
      Local time: Thu 2023-06-29 16:31:49 UTC
  Universal time: Thu 2023-06-29 16:31:49 UTC
        RTC time: n/a
       Time zone: n/a (UTC, +0000)
 Network time on: no
NTP synchronized: no
 RTC in local TZ: no

real    0m0.595s
user    0m0.019s
sys     0m0.037s
root@BMC:~# busctl |cat |grep "org.freedesktop.timedate1"
org.freedesktop.timedate1                         6720 systemd-timedat root             :1.3480       systemd-timedated.service -          -
root@BMC:~#
root@BMC:~# systemctl --version
systemd 234
+PAM -AUDIT -SELINUX -IMA -APPARMOR -SMACK +SYSVINIT -UTMP -LIBCRYPTSETUP -GCRYPT -GNUTLS +ACL +XZ -LZ4 -SECCOMP +BLKID -ELFUTILS +KMOD -IDN2 -IDN default-hierarchy=hybrid
root@BMC:~#

root@BMC:~# time
.bash_history                bmcweb_persistent_data.json
root@BMC:~# time timedatectl
               Local time: Sun 2020-03-22 18:57:04 CST
           Universal time: Sun 2020-03-22 10:57:04 UTC
                 RTC time: n/a
                Time zone: Asia/Shanghai (CST, +0800)
System clock synchronized: no
              NTP service: inactive
          RTC in local TZ: no

real    0m4.428s
user    0m0.036s
sys     0m0.039s
root@BBMC:~#  busctl |cat |grep "org.freedesktop.timedate1"
org.freedesktop.timedate1                        2964 systemd-timedat root            :1.143        systemd-timedated.service                                -       -
root@BMC:~# time timedatectl
               Local time: Sun 2020-03-22 18:57:09 CST
           Universal time: Sun 2020-03-22 10:57:09 UTC
                 RTC time: n/a
                Time zone: Asia/Shanghai (CST, +0800)
System clock synchronized: no
              NTP service: inactive
          RTC in local TZ: no

real    0m0.541s
user    0m0.036s
sys     0m0.039s
root@BMC:~#
root@BMC:~# systemctl --version
systemd 250 (250.5+)
+PAM -AUDIT -SELINUX -APPARMOR -IMA -SMACK +SECCOMP -GCRYPT -GNUTLS -OPENSSL -ACL +BLKID -CURL -ELFUTILS -FIDO2 -IDN2 -IDN -IPTC +KMOD -LIBCRYPTSETUP +LIBFDISK -PCRE2 -PWQUALITY -P11KIT -QRENCODE -BZIP2 -LZ4 -XZ -ZLIB +ZSTD -BPF_FRAMEWORK -XKBCOMMON -UTMP +SYSVINIT default-hierarchy=hybrid
root@BMC:~#

jack206, 2023-11-20:
[log.zip](https://github.com/openbmc/openbmc/files/13411677/log.zip)
I execute timedatectl in systemd（version 234）,It only takes 0.5s，but in version 250,it takes almost 5s.Does anyone know why？

jack206, 2023-11-20:
I can tell it has nothing to do with cpu usage:
em: 374108K used, 52452K free, 85424K shrd, 68908K buff, 251484K cached
CPU:  48% usr  23% sys   0% nic  28% idle   0% io   0% irq   0% sirq
Load average: 0.47 0.29 0.27 1/92 1985
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 1984     1 root     S     7924   2%  25% /lib/systemd/systemd-timedated

geissonator, 2023-11-27:
>  I execute timedatectl in systemd（version 234）,It only takes 0.5s，but in version 250,it takes almost 5s.Does anyone know why？

Sounds like more of a systemd bug then?

Also please wrap copy/pasted output text with the 3 backticks. It make it much easier to read and prevents github from trying to format it for you.

```
i.e. this is text wrapped in a triple backtick
```

I checked my system (AST2600, p10bmc machine) and it seems fine here with systemd 254:
```
# systemctl --version
systemd 254 (254.4^)
+PAM -AUDIT -SELINUX -APPARMOR -IMA -SMACK +SECCOMP -GCRYPT -GNUTLS -OPENSSL -ACL +BLKID -CURL -ELFUTILS -FIDO2 -IDN2 -IDN -IPTC -KMOD -LIBCRYPTSETUP +LIBFDISK -PCRE2 -PWQUALITY -P11KIT -QRENCODE -TPM2 -BZIP2 -LZ4 -XZ -ZLIB +ZSTD -BPF_FRAMEWORK -XKBCOMMON -UTMP +SYSVINIT default-hierarchy=unified

# time timedatectl
               Local time: Mon 2023-11-27 20:59:06 UTC
           Universal time: Mon 2023-11-27 20:59:06 UTC
                 RTC time: Mon 2023-11-27 20:59:06
                Time zone: UTC (UTC, +0000)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no

real    0m0.036s
user    0m0.000s
sys     0m0.017s
```

jack206, 2023-11-30:
root@BMC:~# devmem  0x1E785004 32 0x014FB180
root@BMC:~# devmem  0x1E785008 32 0x4755
root@BMC:~# devmem  0x1E785000
0x014FB180
root@BMC:~#
root@BMC:~# reboot

Remote side unexpectedly closed network connection
root@TCS9500:~# devmem  0x1E785000
0x0001F400
root@BMC:~#

mdmillerii, 2023-11-30:
I notice a possibly significant difference, the slow time was reported with the clock not synchronized.

jack206, 2023-11-30:
> Mon

Can you be more specific？

jack206, 2023-11-30:
> I notice a possibly significant difference, the slow time was reported with the clock not synchronized.

Can you be more specific？

edtanous, 2024-01-05:
Looks like the same root cause as https://github.com/openbmc/bmcweb/issues/264

jack206, 2024-03-22:
It seems systemed(253) also have this case.
root@BMC:~# systemctl --version
systemd 253 (253.1^)
+PAM -AUDIT -SELINUX -APPARMOR -IMA -SMACK +SECCOMP -GCRYPT -GNUTLS -OPENSSL -ACL +BLKID -CURL -ELFUTILS -FIDO2 -IDN2 -IDN -IPTC -KMOD -LIBCRYPTSETUP +LIBFDISK -PCRE2 -PWQUALITY -P11KIT -QRENCODE -TPM2 -BZIP2 -LZ4 -XZ -ZLIB +ZSTD -BPF_FRAMEWORK -XKBCOMMON -UTMP -SYSVINIT default-hierarchy=unified
root@BMC:~# time timedatectl 
               Local time: Wed 2024-03-20 17:08:06 CST
           Universal time: Wed 2024-03-20 09:08:06 UTC
                 RTC time: n/a
                Time zone: Asia/Shanghai (CST, +0800)
System clock synchronized: no
              NTP service: active
          RTC in local TZ: no

real    0m2.136s
user    0m0.032s
sys     0m0.033s
    TimeSyncMethod ync_method xyz.openbmc_project.Time.Synchronization           
s "xyz.openbmc_project.Time.Synchronization.Method.NTP"
nbmc_project/time/sync_method xyz.openbmc_project.Time.Synchronization  \xyz/open
anual         TimeSyncMethod s xyz.openbmc_project.Time.Synchronization.Method.Ma
root@BMC:~# time timedatectl 
Failed to query server: Remote peer disconnected

real    0m17.219s
user    0m0.036s
sys     0m0.025s
root@B6580-48CQ8QC:~#

matthewfischer, 2024-12-18:
I am not seeing the delay on systemd 255. I see above that yours is failing to contact the remote ntp server - could that explain the delay?  I tried setting my config to non-existent servers but this didn't reproduce like it did for you.