openbmc/openbmc #3969: xyz.openbmc_project.State.Host@0.service: Two services allocated for the same bus name

IgorLinux1990, 2023-11-24:
**Note on Bugs**
Revision commit 6838acd66f1fd15761282c181e689cc334dc635c 
https://github.com/openbmc/openbmc/commit/6838acd66f1fd15761282c181e689cc334dc635c

static.mtd image built for x11spi hardware
Boot process freezes with following messages:

Welcome to Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.15.0-dev!
[    4.371518] systemd[1]: Hostname set to <x11spi>.
[    4.415243] systemd[1]: Initializing machine ID from random generator.
[    4.680825] systemd[1]: Using hardware watchdog 'aspeed_wdt', version 0, device /dev/watchdog0
[    4.682584] systemd[1]: Watchdog running with a timeout of 2min.
[    4.683633] systemd[1]: Watchdog: reading from /sys/dev/char/247:0/pretimeout_governor
[    4.685127] systemd[1]: Watchdog: failed to read pretimeout governor: No such file or directory
[    4.685911] systemd[1]: Watchdog: setting pretimeout_governor to 'panic' via '/sys/dev/char/247:0/pretimeout_governor'
[    4.687274] systemd[1]: Failed to set pretimeout_governor to 'panic': No such file or directory
[    4.687930] systemd[1]: Failed to set watchdog pretimeout governor to 'panic', ignoring: No such file or directory
[    6.602932] systemd[1]: Failed to put bus name to hashmap: File exists
[    6.603805] systemd[1]: xyz.openbmc_project.State.Host@0.service: Two services allocated for the same bus name xyz.openbmc_project.State.Host0, refusing operation.
[    8.822291] systemd[1]: /usr/lib/systemd/system/phosphor-ipmi-net@.socket:6: Invalid interface name, ignoring: sys-subsystem-net-devices-%i.device
[    8.980274] systemd[1]: Failed to isolate default target: Unit xyz.openbmc_project.State.Host@0.service failed to load properly, please adjust/correct and reload service manager: File exists
[!!!!!!] Failed to isolate default target.
[    9.138925] systemd[1]: Freezing execution.

**Unexpected behavior you saw**
OS is freeze.

**Expected behavior**
Boot success till login

**To Reproduce**
Steps to reproduce the behavior:

    source ./setup x11spi

    bitbake obmc-phosphor-image

   ./qemu-system-arm -M supermicrox11-bmc -nic user -nographic -drive file=flash-x11spi,if=mtd,format=raw

    See error
Welcome to Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.15.0-dev!
[    4.371518] systemd[1]: Hostname set to <x11spi>.
[    4.415243] systemd[1]: Initializing machine ID from random generator.
[    4.680825] systemd[1]: Using hardware watchdog 'aspeed_wdt', version 0, device /dev/watchdog0
[    4.682584] systemd[1]: Watchdog running with a timeout of 2min.
[    4.683633] systemd[1]: Watchdog: reading from /sys/dev/char/247:0/pretimeout_governor
[    4.685127] systemd[1]: Watchdog: failed to read pretimeout governor: No such file or directory
[    4.685911] systemd[1]: Watchdog: setting pretimeout_governor to 'panic' via '/sys/dev/char/247:0/pretimeout_governor'
[    4.687274] systemd[1]: Failed to set pretimeout_governor to 'panic': No such file or directory
[    4.687930] systemd[1]: Failed to set watchdog pretimeout governor to 'panic', ignoring: No such file or directory
[    6.602932] systemd[1]: Failed to put bus name to hashmap: File exists
[    6.603805] systemd[1]: xyz.openbmc_project.State.Host@0.service: Two services allocated for the same bus name xyz.openbmc_project.State.Host0, refusing operation.
[    8.822291] systemd[1]: /usr/lib/systemd/system/phosphor-ipmi-net@.socket:6: Invalid interface name, ignoring: sys-subsystem-net-devices-%i.device
[    8.980274] systemd[1]: Failed to isolate default target: Unit xyz.openbmc_project.State.Host@0.service failed to load properly, please adjust/correct and reload service manager: File exists
[!!!!!!] Failed to isolate default target.
[    9.138925] systemd[1]: Freezing execution.

OpenBMC Information:
Revision https://github.com/openbmc/openbmc/commit/6838acd66f1fd15761282c181e689cc334dc635c master branch

Additional context
QEMU emulator version 8.1.2

Probably this issue related to [phosphor-state-manager changes on Apr 22, 2022](https://github.com/openbmc/openbmc/issues/3853)

**Additional context**
Add any other context about the problem here.

geissonator, 2023-11-27:
Sounds like you may have both x86-power-control and phosphor-state-manager (-host) in your image? If you want to use x86-power-control then you add something like this to your machine conf:
```
VIRTUAL-RUNTIME_obmc-host-state-manager ?= "x86-power-control"
```

Of if you don't need any of the phosphor-state-manager packages you can completely remove it.

IgorLinux1990, 2023-12-01:
Yes, I added the specified lines to the configuration file(/meta-supermicro/conf/machine/include/supermicro.inc):

# since we use x86-power-control service, we don't need below three services
VIRTUAL-RUNTIME_obmc-host-state-manager = "x86-power-control"
VIRTUAL-RUNTIME_obmc-chassis-state-manager = "x86-power-control"
VIRTUAL-RUNTIME_obmc-discover-system-state = "x86-power-control"

PREFERRED_PROVIDER_virtual/obmc-chassis-mgmt = "packagegroup-supermicro-apps"
PREFERRED_PROVIDER_virtual/obmc-fan-mgmt = "packagegroup-supermicro-apps"
PREFERRED_PROVIDER_virtual/obmc-flash-mgmt = "packagegroup-supermicro-apps"
PREFERRED_PROVIDER_virtual/obmc-system-mgmt = "packagegroup-supermicro-apps"
PREFERRED_PROVIDER_virtual/obmc-inventory-data ?= "${VIRTUAL-RUNTIME_skeleton_w>
PREFERRED_PROVIDER_virtual/obmc-host-ipmi-hw ?= "phosphor-ipmi-kcs"

And it works! But:
root@x11spi:~# systemctl status xyz.openbmc_project.Chassis.Control.Power@0.service 
x xyz.openbmc_project.Chassis.Control.Power@0.service - Intel Power Control for the Host 0
     Loaded: loaded (/usr/lib/systemd/system/xyz.openbmc_project.Chassis.Control.Power@.service; static)
     Active: failed (Result: exit-code) since Fri 2023-12-01 12:57:54 UTC; 2min 16s ago
   Duration: 186ms
    Process: 332 ExecStart=/usr/bin/power-control 0 (code=exited, status=255/EXCEPTION)
   Main PID: 332 (code=exited, status=255/EXCEPTION)
        CPU: 59ms

Dec 01 12:57:54 x11spi systemd[1]: xyz.openbmc_project.Chassis.Control.Power@0.service: Scheduled restart job, restart counter is at 2.
Dec 01 12:57:54 x11spi systemd[1]: xyz.openbmc_project.Chassis.Control.Power@0.service: Start request repeated too quickly.
Dec 01 12:57:54 x11spi systemd[1]: xyz.openbmc_project.Chassis.Control.Power@0.service: Failed with result 'exit-code'.
Dec 01 12:57:54 x11spi systemd[1]: Failed to start Intel Power Control for the Host 0.

root@x11spi:~# systemctl status phosphor-ipmi-kcs@ipmi-kcs3.service
x phosphor-ipmi-kcs@ipmi-kcs3.service - Phosphor IPMI KCS DBus Bridge
     Loaded: loaded (/usr/lib/systemd/system/phosphor-ipmi-kcs@.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Fri 2023-12-01 12:57:55 UTC; 3min 11s ago
    Process: 339 ExecStart=/usr/libexec/kcsbridged -c ipmi-kcs3 (code=exited, status=1/FAILURE)
   Main PID: 339 (code=exited, status=1/FAILURE)
        CPU: 31ms

Dec 01 12:57:55 x11spi systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Scheduled restart job, restart counter is at 2.
Dec 01 12:57:55 x11spi systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Start request repeated too quickly.
Dec 01 12:57:55 x11spi systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Failed with result 'exit-code'.
Dec 01 12:57:55 x11spi systemd[1]: Failed to start Phosphor IPMI KCS DBus Bridge.

root@x11spi:~# bmcutil status
-sh: bmcutil: command not found

I'm thinking of building an old version of the x11 spi revision: commit dac14296f909ccc6b8fa8b2588c3b2629e22e0d6

IgorLinux1990, 2023-12-11:
I also found the TX pin on the x11SPI-TF board, I think this information will also be useful.
[image]

vlozinin, 2025-05-26:
klknlk