openbmc/openbmc #3971: phosphor-debug-collector causes kernel out of memory issue

QuangNguyen2016, 2023-11-27:
**Bug description**
When receiving an elog event, phosphor-dump-manager will fork new processes to collect data and compress data for use when needed. But if an application somehow got issue and flood the elog events to phosphor-dump-manager, it will end up creating a fork bomb which creates a significant large amount of processes and consume all the system memory. It causes the kernel out of memory mechanism is triggered to kill processes and free up memory.

I do not know if there is any mechanism to avoid phosphor-dump-manager to be flooded with elog events to avoid the out of memory issue.

I see this commit enable receiving elog events
https://github.com/openbmc/phosphor-debug-collector/commit/e4350f937f4cc147ed53711f3b5021bfd85392f4

**Unexpected behavior you saw**
When phosphor-dump-manager is flooded with elog events, the BMC console will print out kernel out of memory issue.

**Expected behavior**
When receiving too many elog events, it should not cause the kernel out of momory to happen.
There should be a mechanism to avoid phosphor-dump-manager stop receiving elog event when it is flooded with events.

**To Reproduce**
Steps to reproduce the behavior:
1. Write a script which continuously sending elog events
e.x: busctl call xyz.openbmc_project.Logging \
 /xyz/openbmc_project/logging \
 xyz.openbmc_project.Logging.Create Create ssa{ss} \
 xyz.openbmc_project.Common.Error.InternalFailure \
 xyz.openbmc_project.Logging.Entry.Level.Error 0
2. Wait for several minutes and check BMC console to see the kernel out of memory is reported
The sample  console output when issue happens:

```
2023-11-21T09:18:17.155729 [  271.724780] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),task=xz,pid=14973,uid=0
2023-11-21T09:18:17.164564 [  271.734018] Out of memory: Killed process 14973 (xz) total-vm:97384kB, anon-rss:19968kB, file-rss:128kB, shmem-rss:0kB, UID:0 pgtables:36kB oom_score_adj:0
2023-11-21T09:18:17.164564 [  279.816325] xz invoked oom-killer: gfp_mask=0x140cc0(GFP_USER|__GFP_COMP), order=0, oom_score_adj=0
2023-11-21T09:18:25.240025 [  279.826554] CPU: 0 PID: 16999 Comm: xz Not tainted 6.5.4-8dbdf5a-g8dbdf5a0e362 #1
2023-11-21T09:18:25.277925 [  279.834923] Hardware name: Generic DT based system
2023-11-21T09:18:25.277925 [  279.840285]  unwind_backtrace from show_stack+0x18/0x1c
2023-11-21T09:18:25.277925 [  279.846148]  show_stack from dump_stack_lvl+0x40/0x4c
2023-11-21T09:18:25.277925 [  279.851806]  dump_stack_lvl from dump_header+0x6c/0x2d0
2023-11-21T09:18:25.277925 [  279.857655]  dump_header from oom_kill_process+0x484/0x494
2023-11-21T09:18:25.294618 [  279.863798]  oom_kill_process from out_of_memory+0xdc/0x494
2023-11-21T09:18:25.294618 [  279.870056]  out_of_memory from __alloc_pages+0xb78/0xe50
2023-11-21T09:18:25.294618 [  279.876109]  __alloc_pages from do_read_cache_folio+0xe4/0x1f8
2023-11-21T09:18:25.294618 [  279.882664]  do_read_cache_folio from read_cache_page+0x20/0x48
2023-11-21T09:18:25.325820 [  279.889329]  read_cache_page from page_get_link+0x34/0x104
2023-11-21T09:18:25.325820 [  279.895517]  page_get_link from ovl_get_link+0x38/0x4c
2023-11-21T09:18:25.325820 [  279.901328]  ovl_get_link from step_into+0x59c/0x690
2023-11-21T09:18:25.325820 [  279.906941]  step_into from path_openat+0x120/0xe80
2023-11-21T09:18:25.325820 [  279.912447]  path_openat from do_filp_open+0x98/0x134
2023-11-21T09:18:25.346441 [  279.918124]  do_filp_open from do_sys_openat2+0xbc/0xf0
2023-11-21T09:18:25.346441 [  279.923980]  do_sys_openat2 from sys_openat+0x80/0xcc
2023-11-21T09:18:25.346441 [  279.929662]  sys_openat from ret_fast_syscall+0x0/0x54
2023-11-21T09:18:25.383667 [  279.935430] Exception stack(0xc04a1fa8 to 0xc04a1ff0)
2023-11-21T09:18:25.383667 [  279.941080] 1fa0:                   00452f58 7ee3ce1c ffffff9c 76f86340 000a0000 00000000
2023-11-21T09:18:25.383667 [  279.950220] 1fc0: 00452f58 7ee3ce1c 00000000 00000142 00000000 00000000 7ee3ce1c 7ee3cdd4
2023-11-21T09:18:25.383667 [  279.959353] 1fe0: 76f86352 7ee3cd98 76f8ce9c 76fa49e4
2023-11-21T09:18:25.409352 [  279.965221] Mem-Info:
2023-11-21T09:18:25.409352 [  279.967801] active_anon:16275 inactive_anon:211765 isolated_anon:0
2023-11-21T09:18:25.409352 [  279.967801]  active_file:3 inactive_file:225 isolated_file:231
2023-11-21T09:18:25.409352 [  279.967801]  unevictable:0 dirty:0 writeback:0
2023-11-21T09:18:25.409352 [  279.967801]  slab_reclaimable:2849 slab_unreclaimable:4922
2023-11-21T09:18:25.409352 [  279.967801]  mapped:576 shmem:19056 pagetables:2652
2023-11-21T09:18:25.430198 [  279.967801]  sec_pagetables:0 bounce:0
2023-11-21T09:18:25.430198 [  279.967801]  kernel_misc_reclaimable:0
2023-11-21T09:18:25.430198 [  279.967801]  free:3108 free_pcp:0 free_cma:116
2023-11-21T09:18:25.430198 [  280.011053] Node 0 active_anon:65100kB inactive_anon:847060kB active_file:12kB inactive_file:1032kB unevictable:0kB isolated(anon):0kB isolated(file):772kB mapped:2304kB dirty:0kB writeback:0kB shmem:76224kB writeback_tmp:0kB kernel_stack:3528kB pagetables:10608kB sec_pagetables:0kB all_unreclaimable? no
2023-11-21T09:18:25.468141 [  280.041221] Normal free:12432kB boost:9028kB min:13040kB low:14044kB high:15048kB reserved_highatomic:0KB active_anon:65100kB inactive_anon:846124kB active_file:88kB inactive_file:1224kB unevictable:0kB writepending:0kB present:1032192kB managed:1008060kB mlocked:0kB bounce:0kB free_pcp:60kB local_pcp:0kB free_cma:464kB
2023-11-21T09:18:25.490399 [  280.072956] lowmem_reserve[]: 0 0 0
2023-11-21T09:18:25.490399 [  280.076878] Normal: 385*4kB (UMEC) 676*8kB (UMEC) 196*16kB (UMEC) 53*32kB (UMEC) 18*64kB (UME) 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 12932kB
2023-11-21T09:18:25.505935 [  280.092546] 19514 total pagecache pages
2023-11-21T09:18:25.505935 [  280.096837] 258048 pages RAM
2023-11-21T09:18:25.505935 [  280.100084] 0 pages HighMem/MovableOnly
2023-11-21T09:18:25.527088 [  280.104363] 6033 pages reserved
2023-11-21T09:18:25.527088 [  280.107871] 24576 pages cma reserved
```

ojayanth, 2023-11-27:
Similar issue fixed for user initiated type dumps.
 https://github.com/openbmc/phosphor-debug-collector/commit/73953b8f8db9a8e8e35ea58f0b8312d823a4a65b.

Currently dump manager doesn't have generic mechanism to stop the dump request for parallel dump requests.  The approach used in user initiated dump requests to ignore the dump request, incase another user initiated dump request is in progress. This approach can't use he because elog type requests captures error specific dumps. 

Just wondering this issue reported in test environment or real production scenario?

Better to move this issue under #phosphor-debug-collector repository from tracking point.

QuangNguyen2016, 2023-11-28:
@ojayanth it happened on our system, I just wrote the test script to demo the issue and make it happen easier. In out system, the phosphor-fru-fault-monitor somehow got issue and keep sending elog events to phosphor-dump-manager. I believe everyone will soon encounter the same issue.

NguyenTanNhutQuang, 2023-12-01:
I added a patch to fix this here
https://gerrit.openbmc.org/c/openbmc/phosphor-debug-collector/+/68029