openbmc/openbmc #3974: MCTP overPCIe VDM

Kendidi, 2023-12-06:
I wonder from Linux host OS, how to send **SPDM via MCTP over PCIe VDM** messages to PCIe device (plugged to a PCIe slot on motherboard) through BMC?  Thanks.

mdmillerii, 2023-12-06:
Do you have documentation how to send and receive VDM packets from the host bridge?   This will be the first step.

williamspatrick, 2023-12-06:
This is not a support channel. Please honor the hints in the issue templates.

closed 2023-12-06 by williamspatrick

Kendidi, 2023-12-06:
> Do you have documentation how to send and receive VDM packets from the host bridge? This will be the first step.

@mdmillerii I do not.   Where can I find those documentation?  Thanks.

williamspatrick, 2023-12-06:
> > This is not a support channel. Please honor the hints in the issue templates.
> 
> @williamspatrick Where should I post similar question then? Thanks.

please read the following:
https://github.com/openbmc/openbmc/issues/new?assignees=&labels=&projects=&template=question.md&title=