openbmc/openbmc #3978: Ipmitool power off /on command not working

vijaysai, 2023-12-16:
I tried openbmc Image compiled for AST2600, In BMC shell if I type ipmitool  chassis power status -> the output is showing system power is on.

Whenever I try to issue ipmitool chassis power off -> I am seeing o/p in serial console as System is Power Down/off, whereas the system is not off, even in Journalctl  -> the output shows that power-control application is able to change the transition to power off,  but there is no impact.

Power-control, is trying to assert POWER_OUT GPIO pin with value 0, can anyone guide me with the code flow for debugging this issue.

Please Guide me for understanding how a GPIO pin change, can switch off chassis or something else is present which I am missing.

Kindly help me in understanding the code flow, where Ipmitool interacts with ipmid and then with power-control application, and then    no effect is observed.

williamspatrick, 2023-12-16:
You've not given any information about what system you are building for, how you've configured it, what commit of OpenBMC you are using. And you've ignored the plain words that we do not give support over GitHub issues. Please follow the support channels identified.

closed 2023-12-16 by williamspatrick

vijaysai, 2023-12-17:
I am trying to flash OpenBmc Image in BMC flash chip for one of the server board, which uses AST2600, I just want to know the flow