openbmc/openbmc #3983: OpenBMC-2.14.0 Failed to run evb-ast2600 image in QEMU

gavinux1, 2024-01-25:
I built evb-ast2600 image and run it in qemu like this:
. setup evb-ast2600
bitbake obmc-phosphor-image
cp tmp/deploy/images/evb-ast2600/obmc-phosphor-image-evb-ast2600.static.mtd .

#./qemu-system-arm \
./tmp/work/evb_ast2600-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/recipe-sysroot-native/usr/bin/qemu-system-arm \
    -m 256 -M bletchley-bmc -nographic \
    -drive file=./obmc-phosphor-image-evb-ast2600.static.mtd,format=raw,if=mtd \
    -global driver=aspeed.ibt,property=chardev,value=ipmi0 \
    -chardev socket,id=ipmi0,host=localhost,port=9002,ipv4=on,server=on,wait=off \
    -net nic \
    -net user,hostfwd=:127.0.0.1:2222-:22,hostfwd=:127.0.0.1:2443-:443,hostfwd=udp:127.0.0.1:2623-:623,hostname=qemu
qemu-system-arm: failed to read the initial flash content

If I use qemu-system-arm from https://jenkins.openbmc.org/job/latest-qemu-x86/lastSuccessfulBuild/artifact/qemu/build/qemu-system-arm
I get this message
qemu-system-arm: device requires 134217728 bytes, block backend provides 67108864 bytes

Now I changed meta-evb/meta-evb-aspeed/meta-evb-ast2600/conf/machine/evb-ast2600.conf
# AST2600 EVB has 64MB SPI flash
FLASH_SIZE = "65536"
#FLASH_SIZE = "131072"

When run with qemu, I get this endless messge
qemu-system-arm: warning: nic ftgmac100.0 has no peer
qemu-system-arm: warning: nic ftgmac100.1 has no peer
qemu-system-arm: warning: nic ftgmac100.3 has no peer

U-Boot SPL 2019.04 (Mar 10 2023 - 01:54:10 +0000)

U-Boot SPL 2019.04 (Mar 10 2023 - 01:54:10 +0000)
already initialized,
U-Boot SPL 2019.04 (Mar 10 2023 - 01:54:10 +0000)
already initialized,

BTW, the openbmc built qemu version is 7,2.0, the dowloaded qemu version is  8.1.50 (v8.0.0-4123-gcc9c5a7784)

williamspatrick, 2024-01-25:
> Now I changed meta-evb/meta-evb-aspeed/meta-evb-ast2600/conf/machine/evb-ast2600.conf
#FLASH_SIZE = "65536"
FLASH_SIZE = "131072"

Please change this back and get the results.  

You need to pad the image out from bitbake to 128M for QEMU instead.

gavinux1, 2024-01-25:
I just change it locally for a quick test run
I pad the image with
dd if=/dev/zero of=obmc-phosphor-image-evb-ast2600.static.mtd bs=1 count=67108864 seek=67108864

**I still get the same endless output from QEMU:**
U-Boot SPL 2019.04 (Mar 10 2023 - 01:54:10 +0000)
already initialized,

gavinux1, 2024-01-26:
I grep  all of the u-boot directory in build/evb-ast2600 folder recursively, but can't find out where the "already initialized." is printed out
build/evb-ast2600$ cat search.sh
for d in `find . -type d -name *[uU]-[bB]oot*`
do
    pushd $d
    grep -r "already initialized\." *
    popd
done

All it finds out are "*****  already initialized." in m68k/cpu/mcf52x2/cpu_init.c. It is not the one I am trying to find out.

gavinux1, 2024-01-26:
Although I didn't find out where the "already initialized." is printed out, I find out why it stuck in booting.
The -m 256 is too small, give -m 1024 will boot normally.

gavinux1, 2024-01-26:
resolved by my self

closed 2024-01-26 by gavinux1

xcode47x, 2024-11-08:
> resolved by my self

Hello,

How please ?