openbmc/openbmc #3984: how to get elog param from dbus call

writePerfectCode, 2024-02-07:
If dbus service trigger an elog such as: elog(NotAllowedReason("xxx")); in dbus method call
how to get the reason "xxx" from bmcweb ?

williamspatrick, 2024-02-07:
Issues are not for support.

closed 2024-02-07 by williamspatrick

writePerfectCode, 2024-02-09:
> Issues are not for support.

I create this from "Issue: Question / Support"

https://github.com/openbmc/openbmc/issues/new?assignees=&labels=&projects=&template=question.md&title=

If not from here, where should from ?

amboar, 2024-02-09:
> > Issues are not for support.
> 
> I create this from "Issue: Question / Support"
> 
> https://github.com/openbmc/openbmc/issues/new?assignees=&labels=&projects=&template=question.md&title=

... and if you read the text of that template, it tells you not to create the issue :slightly_smiling_face: 

You should reach out either by the mailing list or on discord, as documented here:

https://github.com/openbmc/docs/?tab=readme-ov-file#contact