openbmc/openbmc #3990: Why can't my browser connect to bmcweb?

abel840, 2024-03-23:
I use the local port (https://127.0.0.1:2443) to connect to qemu's bmcweb, But I can't log in.

Here is the information from bmcweb:
root@evb-ast2600:~# journalctl -u bmcweb.service
Feb 27 17:26:20 evb-ast2600 systemd[1]: Started Start bmcweb server.
Feb 27 17:26:26 evb-ast2600 bmcweb[250]: [ERROR event_service_manager.hpp:1310] inotify_add_watch failed for redfish log file.
Feb 27 17:26:27 evb-ast2600 bmcweb[250]: Checking certs in file /etc/ssl/certs/https/server.pem
Feb 27 17:26:27 evb-ast2600 bmcweb[250]: Error in verifying signature, regenerating
Feb 27 17:26:27 evb-ast2600 bmcweb[250]: Generating new keys
Feb 27 17:26:27 evb-ast2600 bmcweb[250]: Generating EC key
Feb 27 17:26:27 evb-ast2600 bmcweb[250]: Generating x509 Certificate
Feb 27 17:27:35 evb-ast2600 bmcweb[250]: [CRITICAL http_connection.hpp:583] 0x0 Failed to capture connection
Feb 27 17:27:35 evb-ast2600 bmcweb[250]: [CRITICAL http_connection.hpp:583] 0x0 Failed to capture connection
Feb 27 17:28:17 evb-ast2600 bmcweb[250]: [CRITICAL http_connection.hpp:583] 0x0 Failed to capture connection
Feb 27 17:28:17 evb-ast2600 bmcweb[250]: [CRITICAL http_connection.hpp:583] 0x0 Failed to capture connection
Feb 27 17:28:19 evb-ast2600 bmcweb[250]: [CRITICAL http_connection.hpp:583] 0x0 Failed to capture connection
Feb 27 17:28:19 evb-ast2600 bmcweb[250]: [CRITICAL http_connection.hpp:583] 0x0 Failed to capture connection
Feb 27 17:28:20 evb-ast2600 bmcweb[250]: [CRITICAL http_connection.hpp:583] 0x0 Failed to capture connection
Feb 27 17:28:20 evb-ast2600 bmcweb[250]: [CRITICAL http_connection.hpp:583] 0x0 Failed to capture connection
Feb 27 17:28:20 evb-ast2600 bmcweb[250]: [CRITICAL http_connection.hpp:583] 0x0 Failed to capture connection
Feb 27 17:28:29 evb-ast2600 bmcweb[250]: [CRITICAL http_connection.hpp:583] 0x0 Failed to capture connection
Feb 27 17:28:35 evb-ast2600 bmcweb[250]: [CRITICAL http_connection.hpp:583] 0x0 Failed to capture connection

williamspatrick, 2024-03-24:
How did you forward QEMU from port 2443 to bmcweb?

abel840, 2024-03-24:
> How did you forward QEMU from port 2443 to bmcweb?

I use command:
"./qemu-system-arm -m 1024 -M ast2600-evb -nographic \
       -drive file=./obmc-phosphor-image-evb-ast2600.static.mtd,format=raw,if=mtd \
       -net nic \
       -net user,hostfwd=:127.0.0.1:2222-:22,hostfwd=:127.0.0.1:2443-:443,hostfwd=udp:127.0.0.1:2623-:623,hostname=qemu"

Here is the debug information from bmcweb:

Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:403] 0xd7fe50 async_read_header 649 Bytes
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:369] Fetch the client IP address
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:369] Fetch the client IP address
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG authentication.hpp:112] [AuthMiddleware] X-Auth-Token authentication
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG authentication.hpp:130] [AuthMiddleware] Cookie authentication
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG authentication.hpp:288] authHeader=
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG authentication.hpp:96] [AuthMiddleware] Token authentication
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG authentication.hpp:41] [AuthMiddleware] Basic authentication
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:453] Starting quick deadline
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:470] 0xd7fe50 doRead
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:608] 0xd7fe50 timer started
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:481] 0xd7fe50 async_read_some 40 Bytes
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:369] Fetch the client IP address
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [INFO http_connection.hpp:249] Request:  0xd7fe50 HTTP/1.1 POST /login ::ffff:10.0.2.2
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:281] Setting completion handler
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_response.hpp:238] 0xc29728 setting completion handler
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG routing.hpp:669] Matched rule '/login' 5 / 32
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_response.hpp:228] 0xc29728 calling completion handler
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_response.hpp:231] 0xc29728 completion handler was valid
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_response.hpp:75] Moving response containers; this: 0xd83150; other: 0xc29728
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [INFO complete_response_fields.hpp:22] Response:  /login 200
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:548] 0xd7fe50 doWrite
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [INFO http_body.hpp:202] Returning 37 bytes more=false
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_response.hpp:238] 0xd83150 setting completion handler
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:515] 0xd7fe50 async_write 1303 bytes
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:532] 0xd7fe50 Clearing response
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_response.hpp:191] 0xd83150 Clearing response containers
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:392] 0xd7fe50 doReadHeaders
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:80] 0xd3a2b8 Connection open, total 6
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:403] 0xd7fe50 async_read_header 598 Bytes
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:369] Fetch the client IP address
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:369] Fetch the client IP address
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG authentication.hpp:112] [AuthMiddleware] X-Auth-Token authentication
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG authentication.hpp:130] [AuthMiddleware] Cookie authentication
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG authentication.hpp:288] authHeader=
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG authentication.hpp:96] [AuthMiddleware] Token authentication
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG authentication.hpp:41] [AuthMiddleware] Basic authentication
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:453] Starting quick deadline
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:369] Fetch the client IP address
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [INFO http_connection.hpp:249] Request:  0xd7fe50 HTTP/1.1 GET /redfish/v1/AccountService/Accounts/root ::ffff:10.0.2.2
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [WARNING http_connection.hpp:270] Authentication failed
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_response.hpp:75] Moving response containers; this: 0xd83150; other: 0xd83150
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [INFO complete_response_fields.hpp:22] Response:  /redfish/v1/AccountService/Accounts/root 401
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:548] 0xd7fe50 doWrite
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:608] 0xd7fe50 timer started
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [INFO http_body.hpp:202] Returning 37 bytes more=false
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_response.hpp:238] 0xd83150 setting completion handler
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:608] 0xd254b0 timer started
[... cut ...]
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_response.hpp:238] 0xd65080 setting completion handler
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:515] 0xd61d80 async_write 4833 bytes
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:532] 0xd61d80 Clearing response
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_response.hpp:191] 0xd65080 Clearing response containers
Feb 27 17:27:16 evb-ast2600 bmcweb[248]: [DEBUG http_connection.hpp:392] 0xd61d80 doReadHeaders

closed 2024-03-26 by abel840

abel840, 2024-03-26:
> How did you forward QEMU from port 2443 to bmcweb?
It seems like it's because webui-vue doesn't support authentication methods like cookies, sessions, etc., thank you very much!