openbmc/openbmc #3991: Using virtual media by http protocol

scoopex, 2024-03-26:
[According to the design document ](https://github.com/openbmc/docs/blob/master/designs/virtual-media.md), OpenBMC not seems to provide the possibility to mount virtual media images using the http protocol.

In my view, this is a very impractical design decision:

- since in many management networks no DNS resolution is available for various reasons and HTTPS without hostname makes little sense
- because providing an HTTPS server with official certificates is sometimes a cumbersome task in management networks (Lets Encrypt is usually explicitly not wanted because it requires public reachability or DNS records)
- because the validity of the certificate chain deployed on the OpenBMC can expire over time if the hardware is used for several years.

My suggestion would be that you either:
- also allow HTTP (but possibly give the user a hint)
- or introduce a toggle that allows certificate problems to be ignored

williamspatrick, 2024-03-26:
I'm not sure that this is an "issue". I suggest taking this to the mailing list and/or make a proposed change to this document and the code. 

I'm not positive if this design document is even implemented as stated since there was rework being done by developers at Intel. It might be worth contributing to review of their code if this is something you are interested in. 

My opinion is it should be a trivial meson option to enable http if someone decides that "for security reasons" http should be disabled by default so system integrators can make that decision for themselves.

closed 2024-03-26 by williamspatrick

scoopex, 2024-03-26:
I have just noticed this: In the OpenBMC interface this is called "Legacy mode configuration".

It seems that this is exactly how it is implemented, as the help text also describes it this way and the definition of a http:// URL is not accepted.

From my point of view as a user, this is a bug in the design.

You are making it unnecessarily difficult for users in many scenarios and they will not love this behavior if OpenBMC becomes more widespread.

That's why I would advocate reopening the bug :-)

williamspatrick, 2024-03-26:
> From my point of view as a user, this is a bug in the design.

Someone else is going to say that "the BMC should be secure by default" and thus not allow HTTP.  If we enable HTTP by default, they're also going to say that there is a "bug in the design".  You can't make everyone happy, which is why I suggested making this a compile time or runtime option.  Ultimately, the people who wrote that design document and the code behind it _already_ worked to get consensus in the community and if you want to shift that consensus it is going to take more effort than opening GitHub issue.

> You are making it unnecessarily difficult for users in many scenarios and they will not love this behavior if OpenBMC becomes more widespread.

I'm not sure which "users" you're referring to here.  This project isn't like OpenWRT where random people on the internet download it and install it on their existing hardware (and the project probably never can be like that due to needing such detailed hardware information to make a BMC work).  Our users are either:

1. System integrators, who usually contribute to the project directly.
2. Downstream customers of those system integrators.
3. A very small set of people reverse engineering off-the-shelf hardware.

People in group 1 should be more than capable of "fixing" this to their satisfaction by supplying code changes.  People in group 3 either don't care or will figure it out as well.  People in group 2 should be talking to their vendor (the people in group 1) to make sure their requirements are met, right?

The contributors to the project are mostly all employees of group-1 companies and are making code changes to meet the needs of their employer and customers.  There is likely no one contributing to the project that is going to take up this issue just because.