openbmc/openbmc #3992: log repeat print

jack206, 2024-03-29:
**Note on Bugs**
when i use logger command,log records can be recorded normally

**Unexpected behavior you saw**
but  int bmc_debug.log log is recorded twice

**Expected behavior**
log is recorded once

**To Reproduce**
Steps to reproduce the behavior:
1.logger  ccccc
2.tail -f  /var/log/bmc_debug.log

**Screenshots**
If applicable, add screenshots to help explain your problem.
···
root@BMC:~# logger  ccccc
root@BMC:~# tail -f  /var/log/bmc_debug.log
Mar 29 14:19:06.459088 BMC INFO bmc#systemd[1]: Starting Cleanup of Temporary Directories...
Mar 29 14:19:06.454385 BMC INFO bmc#systemd[1]: Starting Cleanup of Temporary Directories...
Mar 29 14:19:06.813770 BMC INFO bmc#systemd[1]: systemd-tmpfiles-clean.service: Deactivated successfully.
Mar 29 14:19:06.821169 BMC INFO bmc#systemd[1]: Finished Cleanup of Temporary Directories.
Mar 29 14:19:06.890257 BMC INFO bmc#systemd[1]: run-credentials-systemd\x2dtmpfiles\x2dclean.service.mount: Deactivated successfully.
Mar 29 14:19:06.813082 BMC INFO bmc#systemd[1]: systemd-tmpfiles-clean.service: Deactivated successfully.
Mar 29 14:19:06.820492 BMC INFO bmc#systemd[1]: Finished Cleanup of Temporary Directories.
Mar 29 14:19:06.889576 BMC INFO bmc#systemd[1]: run-credentials-systemd\x2dtmpfiles\x2dclean.service.mount: Deactivated successfully.
Mar 29 14:20:04.492690 BMC NOTICE bmc#root: ccccc
Mar 29 14:20:04.487283 BMC NOTICE bmc#root[960]: ccccc
^C
root@BMC:~# journalctl -f
Mar 29 14:11:49 BMC phosphor-watchdog[506]: watchdog: disabled
Mar 29 14:11:49 BMC sel-logger[344]: SMSOS enable watchdog countdown 300 seconds None action
Mar 29 14:12:53 BMC root[880]: test
Mar 29 14:13:26 BMC root[891]: test
Mar 29 14:13:35 BMC root[893]: test 111
Mar 29 14:19:06 BMC systemd[1]: Starting Cleanup of Temporary Directories...
Mar 29 14:19:06 BMC systemd[1]: systemd-tmpfiles-clean.service: Deactivated successfully.
Mar 29 14:19:06 BMC systemd[1]: Finished Cleanup of Temporary Directories.
Mar 29 14:19:06 BMC systemd[1]: run-credentials-systemd\x2dtmpfiles\x2dclean.service.mount: Deactivated successfully.
Mar 29 14:20:04 BMC root[960]: ccccc
···

**OpenBMC Information:**
···
root@BMC:~# uname -a 
Linux BMC 5.15.5-ebd46c8 #1 Thu Nov 25 23:12:27 UTC 2021 armv6l GNU/Linux
root@BMC:~# 
root@BMC:~# 
root@BMC:~# cat /etc/os-release 
ID=openbmc-openpower
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.00.0"
VERSION_ID="v1.00.0-2.14.0-fbe6bfde"
VERSION_CODENAME="langdale"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.00.0"
BUILD_ID="20240328072057"
OPENBMC_TARGET_MACHINE="obmc-wb"
EXTENDED_VERSION="2.14.0"
···
 - Any additional patches or changes:

**Additional context**
Add any other context about the problem here.

williamspatrick, 2024-04-02:
What is `/var/log/bmc_debug.log`?  We have no reference to that anywhere in the repository as best I can tell.  Is this a custom rsyslog config you have created in your own tree?

jack206, 2024-04-03:
yes，here is my rsyslog conf：
···
f $programname == "dhclient" then /var/log/dhclient.log
else if $syslogfacility-text == 'local3' then

# Send local3 log messages to the /var/log/console_syslog.log
# limit size to 30M
local3.*          /var/log/console_syslog.log

else
{
    /var/log/bmc_debug.log;LogUtilFileFormat
}

···

jack206, 2024-04-03:
> What is `/var/log/bmc_debug.log`? We have no reference to that anywhere in the repository as best I can tell. Is this a custom rsyslog config you have created in your own tree?

yes，here is my rsyslog conf：
```
if $programname == "dhclient" then /var/log/dhclient.log
else if $syslogfacility-text == 'local3' then

Send local3 log messages to the /var/log/console_syslog.log
limit size to 30M
local3.* /var/log/console_syslog.log

else
{
/var/log/bmc_debug.log;LogUtilFileFormat
}
```

williamspatrick, 2024-04-03:
This is not an openbmc bug.  As you see, the content is only in the journal once. If you need to work out how to configure rsyslog, please work through that project directly.

closed 2024-04-03 by williamspatrick