openbmc/openbmc #3993: Local build with meson subprojects

Kostr, 2024-03-29:
Meson is considered as a priority build system for the OpenBMC projects.
Since meson has some sort of a packaging system all the current repositories provide subproject folders and wrap files for easy local build. However not all projects are built without issues.

As a test I've tried to perform standard build via:
```
meson setup build && cd build && meson compile
```
for most of the OpenBMC projects.

## The following projects were built successfully:
- libpldm
- dbus-sensors
- sdbusplus (meson setup build -Dtests=disabled -Dexamples=disabled)
- phosphor-dbus-interfaces
- obmc-console
- entity-manager
- phosphor-networkd
- phosphor-host-postd
- phosphor-certificate-manager
- phosphor-virtual-sensor
- sdeventplus
- phosphor-watchdog
- phosphor-time-manager
- gpioplus
- kcsbridge
- phosphor-hwmon
- libpeci
- peci-pcie
- phosphor-buttons
- btbridged
- dbus-top

## Failed

### The following projects failed with the message:
```
cereal| Exception: Failed to configure the CMake subproject: Could NOT find Boost (missing: Boost_INCLUDE_DIR serialization)
Subproject subprojects/cereal is buildable: NO (disabling)
```
- phosphor-logging
- phosphor-user-manager
- phosphor-bmc-code-mgmt
- phosphor-state-manager
- phosphor-debug-collector
- phosphor-inventory-manager
- phosphor-post-code-manager
- phosphor-led-manager
- phosphor-fan-presence
- phosphor-settingsd

### bmcweb

Failed because tinyxml2 dependency has a warning and all warning treated as errors:
```
ignoring return value of ‘char* fgets(char*, int, FILE*)’
```

### phosphor-ipmid-host

Problem with a `boost_coroutine` library

### phosphor-sel-logger

`fatal error: boost/algorithm/string.hpp: No such file or directory`

### The following projects failed with the message:
```meson.build:20:12: ERROR: Dependency "boost" not found, tried system```
- smbios-mdr
- phosphor-power
- ipmbbridge
[... 117 lines cut ...]
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 23.04
Release:        23.04
Codename:       lunar

$ gcc --version
gcc (Ubuntu 13.1.0-2ubuntu2~23.04) 13.1.0

$ g++ --version
g++ (Ubuntu 13.1.0-2ubuntu2~23.04) 13.1.0

$ sudo apt-get install libpam0g-dev
$ sudo apt-get install libsystemd-dev
$ sudo apt-get install libdbus-1-dev      # for EM
$ sudo apt-get install libnl-3-dev            # for phosphor-networkd
$ sudo apt-get install libnl-genl-3-dev
$ sudo apt-get install libgpiod-dev         # for phosphor-host-postd
$ sudo apt-get install libssl-dev              # for phosphor-certificate-manager
$ sudo apt-get install libsnmp-dev         # for phosphor-snmp
$ sudo apt-get install libevdev-dev        # for phosphor-gpio-monitor
$ sudo apt-get install libgtest-dev         # for phosphor-time-manager
$ pip3 install jsonschema                       # for EM
$ pip3 install inflection pyyaml mako
```

Kostr, 2024-03-29:
- MERGED `70330: meson: Provide missing wrap files` | https://gerrit.openbmc.org/c/openbmc/phosphor-hwmon/+/70330
- ACTIVE `70346: build: Fix pkg-config generation on local build` | https://gerrit.openbmc.org/c/openbmc/phosphor-snmp/+/70346
- ACTIVE `70342: build: Provide wrap files for the local build` | https://gerrit.openbmc.org/c/openbmc/peci-pcie/+/70342
- ACTIVE `70326: build: Fix requirements for tests and example features` | https://gerrit.openbmc.org/c/openbmc/sdbusplus/+/70326

williamspatrick, 2024-04-02:
Do we really want to provide a _versioned_ boost subproject for every repository?  This seems like vast overkill.  Who is going to maintain these?  Why can't you install boost through your OS?

Kostr, 2024-04-02:
> Do we really want to provide a versioned boost subproject for every repository?

What do you mean about by maintain?

boost.wrap for the `entity-manager` uses boost version 1.71.0 and was commited about 3 years ago, and builds fine without any issues even now
https://github.com/openbmc/entity-manager/blob/b02752f7dee06a22b920eecc26849b5dd6fd1d51/subprojects/boost.wrap#L3

I don't see any issue with the exact version if this version builds correctly. I would even say that I don't really understand the HEAD concept for subprojects. To me it looks like this approach needs much more maintaining since everything could change in the HEAD and break the build.

> Why can't you install boost through your OS?

To me this subprojects concept looks like a python vitual env. I don't want to install things globally, I want to keep them inside this particular project.

And instead of trying to investigate if my package manager has the boost library, how this package is called and so on, executing `meson setup build && cd build && meson compile` seems like more easier thing to do.