openbmc/openbmc #3998: Can't get kernel  on evb-ast2500 board at error memory address

LookOuta, 2024-05-23:
Hello, Could you tell me how to solve this problem ? I met the same problem in using the AST2500.

ast#
DRAM Init-V12-DDR4
0abc1-8Gb-Done
Read margin-DL:0.3901/DH:0.3882 CK (min:0.30)

U-Boot 2019.04 (Feb 29 2024 - 08:14:28 +0000)

SOC : AST2500-A2
RST : WDT3 - Boot
RST : Power On
LPC Mode : SIO:Enable : SuperIO-4e
Eth : MAC0: RMII/NCSI, , MAC1: RGMII,
Model: AST2500 EVB
DRAM: 960 MiB (capacity:1024 MiB, VGA:64 MiB, ECC:off)
MMC: sdhci_slot0@100: 0, sdhci_slot1@200: 1
Loading Environment from SPI Flash... SF: Detected gd25b512m with page size 256 Bytes, erase size 4 KiB, total 64 MiB
*** Warning - bad CRC, using default environment

In: serial@1e784000
Out: serial@1e784000
Err: serial@1e784000
Net:
Warning: ethernet@1e660000 (eth0) using random MAC address - 9a:e0:ce:93:73:c9
eth0: ethernet@1e660000Could not get PHY for eth1: addr 0
Failed to initialize PHY: -19
, eth-1: ethernet@1e680000
Hit any key to stop autoboot: 0
Wrong Image Format for bootm command
ERROR: can't get kernel image
The following is my modification of bootcmd:

ast# printenv bootcmd
bootcmd=bootm 20100000
ast# setenv bootcmd bootm 20080000
ast# boot
Wrong Image Format for bootm command
ERROR: can't get kernel image!

williamspatrick, 2024-05-23:
Submitter ignored issue template.

LookOuta, 2024-05-27:
> Submitter ignored issue template.
Sorry for ignoring issue template. Could you tell me where to find the template. Thank you very much!

closed 2024-05-27 by williamspatrick