openbmc/openbmc #4003: raspberrypi4-64:fitImage-obmc-phosphor-initramfs-raspberrypi4-64-raspberrypi4-64  is too large!

DandanXu1007, 2024-06-19:
> git clone https://github.com/openbmc/openbmc.git
> git checkout -b kirkstone
> export TEMPLATECONF=meta-evb/meta-evb-raspberrypi/conf
> . openbmc-env
> nvim conf/local.conf  change MACHINE ?= "raspberrypi4-64"
> bitbake obmc-phosphor-image

 modifiy  image_types_phosphor.bbclass for my environment:

> diff -u  image_types_phosphor.bbclass image_types_phosphor.bbclass.bk > test.patch && vi test.patch
[image]

DEBUG: Executing python function do_generate_static
DEBUG: Executing shell function do_mk_static_nor_image
131072+0 records in
131072+0 records out
134217728 bytes (134 MB, 128 MiB) copied, 0.151139 s, 888 MB/s
DEBUG: Shell function do_mk_static_nor_image finished
DEBUG: Considering file size=586944 name=/home/dandan/openbmc-kirkstone/openbmc/build/tmp/deploy/images/raspberrypi4-64/u-boot.bin
DEBUG: Spanning start=0K end=1024K
DEBUG: Compare needed=586944 available=1048576 margin=461632
573+1 records in
573+1 records out
586944 bytes (587 kB, 573 KiB) copied, 0.000738432 s, 795 MB/s
DEBUG: Considering file size=10337844 name=/home/dandan/openbmc-kirkstone/openbmc/build/tmp/deploy/images/raspberrypi4-64/fitImage-obmc-phosphor-initramfs-raspberrypi4-64-raspberrypi4-64
DEBUG: Spanning start=1024K end=10240K
DEBUG: Compare needed=10337844 available=9437184 margin=-900660
ERROR: Image '/home/dandan/openbmc-kirkstone/openbmc/build/tmp/deploy/images/raspberrypi4-64/fitImage-obmc-phosphor-initramfs-raspberrypi4-64-raspberrypi4-64' is too large!
DEBUG: Python function do_generate_static finished

williamspatrick, 2024-06-19:
It looks like you modified partition sizes and now the initramfs doesn't fit.

closed 2024-06-19 by williamspatrick

DandanXu1007, 2024-06-20:
Before modified the partition sizes ,i get the below error:
`ERROR: obmc-phosphor-image-1.0-r0 do_generate_static: Image '/home/dandan/openbmc-kirkstone/openbmc/build/tmp/deploy/images/raspberrypi4-64/u-boot.bin' is too large!
ERROR: Logfile of failure stored in: /home/dandan/openbmc-kirkstone/openbmc/build/tmp/work/raspberrypi4_64-openbmc-linux/obmc-phosphor-image/1.0-r0/temp/log.do_generate_static.3486370
ERROR: Task (/home/dandan/openbmc-kirkstone/openbmc/meta-phosphor/recipes-phosphor/images/obmc-phosphor-image.bb:do_generate_static) failed with exit code '1'`
Then, i refer the issue [https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/46507](url) to solve the promble,but it does not work!Then get the error below:
`ERROR: obmc-phosphor-image-1.0-r0 do_generate_static: Image '/home/dandan/openbmc-kirkstone/openbmc/build/tmp/deploy/images/raspberrypi4-64/fitImage-obmc-phosphor-initramfs-raspberrypi4-64-raspberrypi4-64' is too large!
ERROR: Logfile of failure stored in: /home/dandan/openbmc-kirkstone/openbmc/build/tmp/work/raspberrypi4_64-openbmc-linux/obmc-phosphor-image/1.0-r0/temp/log.do_generate_static.3472024
ERROR: Task (/home/dandan/openbmc-kirkstone/openbmc/meta-phosphor/recipes-phosphor/images/obmc-phosphor-image.bb:do_generate_static) failed with exit code '1'`
The method does work in version 2.9.0, but does not work in version kirkstone.