openbmc/openbmc #4005: The password '0penBmc' is causing issues with my typing.

cvam0000, 2024-07-09:
**Issue**: Because of the default password '0penBmc,' I'm struggling to type 'o' or 'O' most of the time. In my daily work, I have to use the OpenBMC distribution and cannot change the password as others are using the same machines too.

**Solution**: Can we think of any other password so that I, and others who are struggling with the same problem, can have an easier time?

Sorry if I'm not following the issue guidelines.

amboar, 2024-07-10:
I expect you're just going to have to adjust; we've had about 8 years of people building habits on that default password. I expect it's not something that's going to change lightly at this point.

cvam0000, 2024-07-10:
Agreed !! It's not an easy thing to change. closing the issue.

closed 2024-07-10 by cvam0000