openbmc/openbmc #4006: After Build Image, there can't export  populate_sdk_ext at v09.02

chuanlinfan, 2024-07-12:
bitbake -c populate_sdk_ext obmc-phosphor-image

erro log as blew:

```text
Initialising tasks: 100% |#####################################################################################################################################################################################################################################| Time: 0:00:03
Sstate summary: Wanted 2 Local 0 Mirrors 0 Missed 2 Current 3301 (0% match, 99% complete)
NOTE: Executing Tasks
ERROR: obmc-phosphor-image-1.0-r0 do_populate_sdk_ext: Error executing a python function in exec_func_python() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_func_python() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:copy_buildsystem(d)
     0003:
File: '/home/chuanlin/ASpeedTech/openbmc/meta/classes-recipe/populate_sdk_ext.bbclass', lineno: 613, function: copy_buildsystem
     0609:    write_unlocked_sigs(d, baseoutpath)
     0610:
     0611:    write_bblayers_conf(d, baseoutpath, sdkbblayers)
     0612:
 *** 0613:    uninative_checksum = copy_uninative(d, baseoutpath)
     0614:
     0615:    write_local_conf(d, baseoutpath, derivative, core_meta_subdir, uninative_checksum)
     0616:
     0617:    prepare_locked_cache(d, baseoutpath, derivative, conf_initpath)
File: '/home/chuanlin/ASpeedTech/openbmc/meta/classes-recipe/populate_sdk_ext.bbclass', lineno: 288, function: copy_uninative
     0284:        uninative_outdir = '%s/downloads/uninative/%s' % (baseoutpath, uninative_checksum)
     0285:        bb.utils.mkdirhier(uninative_outdir)
     0286:        shutil.copy(uninative_file, uninative_outdir)
     0287:
 *** 0288:    return uninative_checksum
     0289:
     0290:def write_local_conf(d, baseoutpath, derivative, core_meta_subdir, uninative_checksum):
     0291:    #check if custome templateconf path is set
     0292:    use_custom_templateconf = d.getVar('SDK_CUSTOM_TEMPLATECONF')
Exception: UnboundLocalError: local variable 'uninative_checksum' referenced before assignment

ERROR: Logfile of failure stored in: /home/chuanlin/ASpeedTech/openbmc/build/evb-ast2600/tmp/work/evb_ast2600-openbmc-linux-gnueabi/obmc-phosphor-image/1.0/temp/log.do_populate_sdk_ext.1803312
ERROR: Task (/home/chuanlin/ASpeedTech/openbmc/meta-phosphor/recipes-phosphor/images/obmc-phosphor-image.bb:do_populate_sdk_ext) failed with exit code '1'
NOTE: Tasks Summary: Attempted 6748 tasks of which 6746 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/chuanlin/ASpeedTech/openbmc/meta-phosphor/recipes-phosphor/images/obmc-phosphor-image.bb:do_populate_sdk_ext
Summary: There was 1 WARNING message.
Summary: There was 1 ERROR message, returning a non-zero exit code.
chuanlin@chuanlin:~/ASpeedTech/openbmc/build/evb-ast2600$ 
[log.do_populate_sdk_ext.1803312.txt](https://github.com/user-attachments/files/16191335/log.do_populate_sdk_ext.1803312.txt)
```

I found a similar post describing this issue, but following the instructions did not solve the problem and instead made more errors

This is link:
https://docs.yoctoproject.org/pipermail/yocto/2018-January/039574.html

williamspatrick, 2024-07-12:
v09.02 sounds like Aspeed versioning and your directory is AspeedTech/openbmc. Please open issue against your vendor (Aspeed) for support on that.

closed 2024-07-12 by williamspatrick

chuanlinfan, 2024-07-15:
Sorry, It's really a mistake，I mentioned the wrong depository. Anyway, thank you for reminding me