openbmc/openbmc #4007: openbmc 2.14 can use BMCWEB with both HTTP and HTTPS?

jack206, 2024-07-26:
**Note on Bugs**
Please note that questions or feature requests should be discussed on the
mailing list or Discord. Please see https://github.com/openbmc/openbmc#contact
for info on how to contact us. Github Issues in openbmc/openbmc are for bugs
only. Please remove this section when opening a bug.

**Unexpected behavior you saw**
A clear and concise description of what unexpectedly happened.

**Expected behavior**
A clear and concise description of what you expected to happen.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**OpenBMC Information:**
 - /etc/os-release:
 - uname -a:
 - Any additional patches or changes:

**Additional context**
Add any other context about the problem here.

williamspatrick, 2024-07-26:
Please don't ignore issue templates.  This is not the appropriate forum for questions.

closed 2024-07-26 by williamspatrick

jack206, 2024-07-29:
> Please don't ignore issue templates. This is not the appropriate forum for questions.
I asked about it in the email and got no reply, so I brought up the bug.