openbmc/openbmc #4009: Master branch can not be built with GCC 14

toshipp, 2024-08-04:
**Unexpected behavior you saw**
The main branch (I used e614ed348455cb8d110d95d97b71149667cb2c05) can not be built with GCC 14.
It failed with the following message.

```
ERROR: cracklib-native-2.9.11-r0 do_compile: oe_runmake failed
ERROR: cracklib-native-2.9.11-r0 do_compile: ExecutionError('/home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/temp/run.do_compile.498947', 1, None, None)
ERROR: Logfile of failure stored in: /home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/temp/log.do_compile.498947
Log data follows:
| DEBUG: Executing python function autotools_aclocals
| DEBUG: SITE files ['endian-little', 'common-linux', 'common-glibc', 'bit-64', 'x86_64-linux', 'common']
| DEBUG: Python function autotools_aclocals finished
| DEBUG: Executing shell function do_compile
| NOTE: make -j 8
| make  all-recursive
| make[1]: Entering directory '/home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/build'
| Making all in m4
| make[2]: Entering directory '/home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/build/m4'
| make[2]: Nothing to be done for 'all'.
| make[2]: Leaving directory '/home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/build/m4'
| Making all in lib
| make[2]: Entering directory '/home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/build/lib'
| /bin/bash ../libtool  --tag=CC   --mode=compile gcc  -DHAVE_CONFIG_H -I. -I../../git/src/lib -I..  -I. -I.. -I../../git/src/lib -DIN_CRACKLIB '-DDEFAULT_CRACKLIB_DICT="/home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/recipe-sysroot-native/usr/share/cracklib/pw_dict"' -Wall -isystem/home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/recipe-sysroot-native/usr/include  -isystem/home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/recipe-sysroot-native/usr/include -O2 -pipe -c -o packlib.lo ../../git/src/lib/packlib.c
| libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I../../git/src/lib -I.. -I. -I.. -I../../git/src/lib -DIN_CRACKLIB -DDEFAULT_CRACKLIB_DICT=\"/home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/recipe-sysroot-native/usr/share/cracklib/pw_dict\" -Wall -isystem/home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/recipe-sysroot-native/usr/include -isystem/home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/recipe-sysroot-native/usr/include -O2 -pipe -c ../../git/src/lib/packlib.c  -fPIC -DPIC -o .libs/packlib.o
| ../../git/src/lib/packlib.c: In function ‘HwmsHostToBigEndian’:
| ../../git/src/lib/packlib.c:153:13: warning: value computed is not used [-Wunused-value]
|   153 |             *pHwms64++;
|       |             ^~~~~~~~~~
| ../../git/src/lib/packlib.c:164:13: warning: value computed is not used [-Wunused-value]
|   164 |             *pHwms32++;
|       |             ^~~~~~~~~~
| ../../git/src/lib/packlib.c: In function ‘HwmsBigEndianToHost’:
| ../../git/src/lib/packlib.c:198:21: warning: operation on ‘pHwms64’ may be undefined [-Wsequence-point]
|   198 |             *pHwms64++ = bswap_64(*pHwms64);
|       |              ~~~~~~~^~
| ../../git/src/lib/packlib.c:209:13: warning: value computed is not used [-Wunused-value]
|   209 |             *pHwms32++;
|       |             ^~~~~~~~~~
| ../../git/src/lib/packlib.c: In function ‘PWClose’:
| ../../git/src/lib/packlib.c:554:40: error: passing argument 1 of ‘HwmsHostToBigEndian’ from incompatible pointer type [-Wincompatible-pointer-types]
|   554 |             HwmsHostToBigEndian(tmp_pwp.hwms, sizeof(tmp_pwp.hwms), en_is32);
|       |                                 ~~~~~~~^~~~~
|       |                                        |
|       |                                        uint32_t * {aka unsigned int *}
| ../../git/src/lib/packlib.c:142:27: note: expected ‘char *’ but argument is of type ‘uint32_t *’ {aka ‘unsigned int *’}
|   142 | HwmsHostToBigEndian(char *pHwms, int nLen,int nBitType)
|       |                     ~~~~~~^~~~~
| make[2]: *** [Makefile:488: packlib.lo] Error 1
| make[2]: Leaving directory '/home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/build/lib'
| make[1]: *** [Makefile:441: all-recursive] Error 1
| make[1]: Leaving directory '/home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/build'
| make: *** [Makefile:371: all] Error 2
| ERROR: oe_runmake failed
| WARNING: /home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/temp/run.do_compile.498947:182 exit 1 from 'exit 1'
| WARNING: Backtrace (BB generated script):
|       #1: bbfatal_log, /home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/temp/run.do_compile.498947, line 182
|       #2: die, /home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/temp/run.do_compile.498947, line 166
|       #3: oe_runmake, /home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/temp/run.do_compile.498947, line 161
|       #4: autotools_do_compile, /home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/temp/run.do_compile.498947, line 156
|       #5: do_compile, /home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/temp/run.do_compile.498947, line 151
|       #6: main, /home/toshi/Dev/openbmc-build/tmp/work/x86_64-linux/cracklib-native/2.9.11/temp/run.do_compile.498947, line 195
ERROR: Task (virtual:native:/home/toshi/Dev/openbmc/meta/recipes-extended/cracklib/cracklib_2.9.11.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 834 tasks of which 828 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  virtual:native:/home/toshi/Dev/openbmc/meta/recipes-extended/cracklib/cracklib_2.9.11.bb:do_compile
Summary: There were 2 ERROR messages, returning a non-zero exit code
```

**Expected behavior**
The build succeeds.

**To Reproduce**
```
bitbake obmc-phosphor-image
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**OpenBMC Information:**
 - /etc/os-release:
 - uname -a:
 - Any additional patches or changes:

**Additional context**
oe-core has a patch to resolve this issue. please backport this.
https://git.openembedded.org/openembedded-core/diff/meta/recipes-extended/cracklib/cracklib/0001-packlib.c-support-dictionary-byte-order-dependent.patch?id=914128f6bd988cde278e087fb9457a0c70c7e5ec

williamspatrick, 2024-08-04:
There is a chain of commits starting here where we are already working through issues related to upstream subtree update, including GCC-14. 

https://gerrit.openbmc.org/c/openbmc/openbmc/+/72220

toshipp, 2024-08-04:
Thank you for your information.

closed 2024-08-04 by toshipp