openbmc/openbmc #4013: Can't login to WebUI with default credentials in dev tutorial

d1nuc0m, 2024-08-28:
**Unexpected behavior you saw**
Cannot login into web UI when following [development tutorial](https://github.com/openbmc/docs/blob/master/development/dev-environment.md).
The same credentials allow for a successful login through SSH or shell in QEMU

**Expected behavior**
Web UI works too

**To Reproduce**
On Ubuntu 22.04 LTS

```bash
# Installed the tools
sudo apt install git python3-distutils gcc g++ make file wget gawk diffstat bzip2 cpio chrpath zstd lz4 bzip2
# Downloaded and compiled OpenBMC without errors
git clone https://github.com/openbmc/openbmc
# Also tried switching to release tag, but result is the same regardless
git checkout 2.14.0
cd openbmc
. setup romulus
# Compiled without errors
bitbake obmc-phosphor-image
# QEMU
wget https://jenkins.openbmc.org/job/latest-qemu-x86/lastSuccessfulBuild/artifact/qemu/build/qemu-system-arm
chmod u+x qemu-system-arm
cp ./tmp/deploy/images/romulus/obmc-phosphor-image-romulus.static.mtd ./
./qemu-system-arm -m 256 -M romulus-bmc -nographic \
    -drive file=./obmc-phosphor-image-romulus.static.mtd,format=raw,if=mtd \
    -net nic \
    -net user,hostfwd=:127.0.0.1:2222-:22,hostfwd=:127.0.0.1:2443-:443,hostfwd=udp:127.0.0.1:2623-:623,hostname=qemu
```

Opening `https://127.0.0.1:2443` and accepting self-signed certificate shows Web UI login form, but after inserting credentials it returns ''Invalid username or password " or "Method Not Allowed"

**OpenBMC Information:**
 - /etc/os-release:
 ```
ID=openbmc-openpower
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.14.0"
VERSION_ID=2.14.0
VERSION_CODENAME="langdale"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.14.0"
BUILD_ID="20240828092807"
OPENBMC_TARGET_MACHINE="romulus"
EXTENDED_VERSION="2.14.0"
```
 - uname -a: `Linux romulus 6.1.15-580639a #1 Thu Apr  6 00:55:09 UTC 2023 armv6l GNU/Linux`
 - Any additional patches or changes: `None`

**Additional context**
Same does not happen when downloading [latest Romolus image](https://jenkins.openbmc.org/job/latest-master/label=docker-builder,target=romulus/lastSuccessfulBuild/artifact/openbmc/build/tmp/deploy/images/romulus/obmc-phosphor-image-romulus.static.mtd).
Maybe it's just something missing in the docs?

d1nuc0m, 2024-08-28:
> You didn't state the actual password you tried. The default starts with a digit zero `0`>

Yes, I tried to use `root`/`0penBmc` as username/password, and it works through SSH or shell in QEMU. It does not work only in web UI

matthewfischer, 2024-12-18:
Working on ast2600 and Romulus as of December 17.

[image]

[image]

closed 2024-12-20 by williamspatrick

d1nuc0m, 2025-02-25:
> Working on ast2600 and Romulus as of December 17.

Issue is related to 2.14.0 (latest stable), the test above is on 2.18.0 if I'm not wrong, could you try with 2.14? Thanks

mdmillerii, 2025-02-25:
The [release process was discontinued](https://github.com/openbmc/docs/commit/d62d386d16b80442274cc997e8e51224af2425e1) approaching [two years ago](https://github.com/openbmc/docs/commit/v2.14) after [no public reports](https://lists.ozlabs.org/pipermail/openbmc/2023-May/thread.html#33390) of testing beyond our [CI](https://github.com/openbmc/docs/tree/master/testing) were made not even [our automated test suite](https://github.com/openbmc/openbmc-test-automation).

For the convenience of downstream users who want to maintain a more stable base we create a [branch](https://github.com/openbmc/openbmc/branches/all) corresponding to the merger of each uptream yocto release with fixes needed to pass our CI .  While we do accept backported patches for our code, it's rare to see them.  Note the commits ahead of master column with the small numbers for these branches

Please test using a recent commit on the master branch and include the sha1 hash you use in your testing.