openbmc/openbmc #4014: bitbake core-image-full-cmdline not able to login with root and 0penBmc

ramthyson, 2024-08-30:
Hi Experts,

1. git clone https://github.com/openbmc/openbmc
2. cd openbmc/
3 . setup evb-ast2600 
bitbake obmc-phosphor-image

Following the above steps, I got an image and flashed the same, but I could not log in using username root and password as 0penBmc.

Error :  
Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) nodistro.0 evb-ast2600 ttyS4

evb-ast2600 login: root
Password:

Login incorrect
evb-ast2600 login: root
Password:

Login incorrect
evb-ast2600 login: root
Password:

Login incorrect
evb-ast2600 login: root
Password:

Login incorrect
evb-ast2600 login: root
Password:

Login incorrect
Maximum numb
Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) nodistro.0 evb-ast2600 ttyS4

evb-ast2600 login:

williamspatrick, 2024-11-02:
The title and the instructions you gave do not match.  `core-image-full-cmdline` is an image-type from Yocto but not one we support.  I have not heard of anyone having issues with passwords with the `obmc-phosphor-image`.  Are you sure the instructions you've given are correct.

matthewfischer, 2024-12-18:
I actually built this image on Sunday (December 15) with the standard bitbake and it's working fine for me, recommend we close this.

```Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) nodistro.0 evb-ast2600 ttyS4

evb-ast2600 login: root
Password:
root@evb-ast2600:~# ls
bmcweb_persistent_data.json
root@evb-ast2600:~#
```

closed 2024-12-18 by williamspatrick