openbmc/openbmc #4015: Sensor added to evb-ast2600 bmc is not visible in webui

Vinay-Rama, 2024-09-03:
Hi i added tmp421 sensor to evb-ast2600. Built an image and launched qemu using generated image. In webui i m not seeing any sensors.

Vinay-Rama, 2024-09-04:
$cat /etc/os-release                             
ID=openbmc-phosphor
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.17.0-dev"
VERSION_ID=2.17.0-dev-415-g8a48c9b4f9-dirty
VERSION_CODENAME="scarthgap"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.17.0-dev"
CPE_NAME="cpe:/o:openembedded:openbmc-phosphor:2.17.0-dev-415-g8a48c9b4f9-dirty"
BUILD_ID="20240903075132"
OPENBMC_TARGET_MACHINE="evb-ast2600"
EXTENDED_VERSION="2.17.0-dev-415-g8a48c9b4f9-dirty"

1. dts
&i2c8 {
	status = "okay";

	lm75@4d {
		compatible = "national,lm75";
		reg = <0x4d>;
	};
	lm75@3d {
		compatible = "national,lm75";
		reg = <0x3d>;
	};
};

&i2c9 {
	status = "okay";
	tmp421@4d {
		compatible = "ti,tmp421";
		reg = <0x4d>;
	};
};
2. phosphor_hwmon_%.bbappend
FILESEXTRAPATHS:prepend := "${THISDIR}/${PN}:"

CHIPS = " \
               bus@1e78a000/i2c-bus@480/lm75@4d \
               bus@1e78a000/i2c-bus@480/lm75@3d \
               bus@1e78a000/i2c-bus@500/tmp421@4d \
               "

ITEMSFMT = "ahb/apb/{0}.conf"
ITEMSFMT:evb-ast2600 = "devices/platform/ahb/ahb--apb/ahb--apb--bus@1e78a000/{0}.conf"
ITEMS = "${@compose_list(d, 'ITEMSFMT', 'CHIPS')}"

ENVS = "obmc/hwmon/{0}"
SYSTEMD_ENVIRONMENT_FILE_${PN}:append:evb-ast2600 = "${@compose_list(d, 'ENVS', 'ITEMS')}"
SYSTEMD_ENVIRONMENT_FILE_${PN}:append:ast2600 = "${@compose_list(d, 'ENVS', 'ITEMS')}"

SYSTEMD_ENVIRONMENT_FILE:${PN} +="obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@480/lm75@4d.conf"
SYSTEMD_ENVIRONMENT_FILE:${PN} +="obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@500/tmp421@4d.conf"

3. conf
cat phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@480/lm75@4d.conf 
LABEL_temp1=ambient
WARNLO_temp1=10000
WARNHI_temp1=40000
CRITHI_temp1=60000
CRITLO_temp1=0

LABEL_temp2=ambient
WARNLO_temp2=10000
WARNHI_temp2=40000
CRITHI_temp2=60000
CRITLO_temp2=0

cat phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@500/tmp421@4d.conf 
LABEL_temp3=ambient
WARNLO_temp3=10000
WARNHI_temp3=40000
CRITHI_temp3=60000
CRITLO_temp3=0

4. my qemu script to launch bmc
./qemu-system-arm -m 512 -M ast2600-evb -nographic \
    -drive file=./obmc-phosphor-image-evb-ast2600-20240903190620.static.mtd,format=raw,if=mtd \
    -net nic \
    -net user,hostfwd=:127.0.0.1:2222-:22,hostfwd=:127.0.0.1:2443-:443,hostfwd=udp:127.0.0.1:2623-:623,hostname=qemu \
    -device tmp105,id=sensor,address=0x50   \
    -device tmp421,id=sensor1,address=0x4d   \
    -qmp unix:$HOME/qmp.sock,server,nowait  \
  =============================================================================
root@evb-ast2600:~# dmesg | grep hwmon
[    0.406328] lm75 8-004d: hwmon0: sensor 'lm75'
root@evb-ast2600:~# journalctl | grep hwmon
Feb 27 17:26:17 evb-ast2600 kernel: lm75 8-004d: hwmon0: sensor 'lm75'
1. journalctl only lm75 is getting detected tmp421 is not reflected?
2. In webui no sensors is detected in sensor list?

amboar, 2024-09-10:
I don't have an answer to your query, but something to be aware of is this change:

https://github.com/openbmc/openbmc/commit/a221e432b030563430d1c1b32a6ebc010d3ba505

Vinay-Rama, 2024-09-15:
I have traced the src code. Before any sensors get populated in webui 1st it will fetch the chassis details and then it will append the chassis id to the url and add sensor (i.e. http:// ip/redfish/v1/chassis/<str>/sensor). But in my case chassis details is getting zero. I doubt because of that sensor details is not present in webpage.

saipavann27, 2025-11-26:
Before Exposing the Sensors into the GUI, We Need confirm the following things Regarding End device presence with the help of the Following command i2cdetect -y <Bus number> (Ex. i2cdetect -y 8). Here it can detect the Slave device, and we need to add the required driver in the kernel.

Note: But in your case, I am able to see the device under the bus 8. But there is no dmesg log for the bus 9 (tmp421). first make sure device is detected under the bus 9.

If Device is successfully detected and registered with the driver our next step to expose the value into the GUI. There is two ways to expose value into the GUI:
                                                      1. Entity Manager. (Dynamically detect the devices)
                                                      2. Phosphor hwmon. (static)

1. Entity Manager: 
                        Entity Manager will detect the devices dynamically based on the Json configuration Like I2c bus number, slave address and Type of the device.

Note: Entity Manager will manager will manage the Hardware tropology. That's why It will create the Inventory path with the chassis. Due to the Redfish Entry will be there for sensors to expose the value for GUI.

                                                  Ex Json:
                                                                
                                                                    {
                                                                          "Name": "EVB Board",
                                                                           "Type": "Board", /* It will create the Entry in Inventory */
                                                                            "Probe": "TRUE",
                                                                       }  

 2. Phosphor hwmon:
                          Phosphor hwmon will detect the device using the hwmon sys entry and create the dbus path. But it will not 
                            create the Inventory path for the Chassis. Due to that Maybe it is not Exposing to GUI.

saipavann27, 2025-11-28:
2. phosphor-hwmon: when we are using this application need to use configuration file and it will expose the value on dbus.
Please find the attached png:
                                                    1. After configuration service will run dynamically.
 
[image]   

[image]

  2. dbus path of phosphor-hwmon application service
  
[image]
[image]
                                                     3.curl command to check chassis entry.

[image]

Note: This application will take care of exposing info upto dbus.it will not take care of create chassis entry for expose info in the GUI.