openbmc/openbmc #4017: ERROR: stdplus-0.1+gitAUTOINC+6047970d6e-r1 do_configure: meson failed

zyt888, 2024-09-25:
when i bitbake stdplus, met this error, how to resolve it? 

I'm doing bitbake under the **romulus** under **version 2.14**
 

../git/meson.build:1:0: ERROR: Executables created by cpp compiler arm-openbmc-linux-gnueabi-g++ -marm -mcpu=arm1176jz-s -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/recipe-sysroot are not runnable.

A full log can be found at /data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/build/meson-logs/meson-log.txt

+ bbfatal_log meson failed

+ '[' -p /data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/temp/fifo.3216531 ']'

ERROR: stdplus-0.1+gitAUTOINC+6047970d6e-r1 do_configure: meson failed
+ printf '%b\0' 'bbfatal_log meson failed'
+ exit 1
+ bb_bash_exit_handler 'exit 1'

+ ret=1

+ set +x

WARNING: /data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/temp/run.do_configure.3216531:173 exit 1 from 'exit 1'
WARNING: Backtrace (BB generated script): 

	#1: bbfatal_log, /data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/temp/run.do_configure.3216531, line 173
	#2: meson_do_configure, /data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/temp/run.do_configure.3216531, line 162

	#3: do_configure, /data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/temp/run.do_configure.3216531, line 148
	#4: main, /data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/temp/run.do_configure.3216531, line 187
ERROR: stdplus-0.1+gitAUTOINC+6047970d6e-r1 do_configure: ExecutionError('/data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/temp/run.do_configure.3216531', 1, None, None)

williamspatrick, 2024-09-25:
This issue is completely garbled formatting and you didn't follow the issue template...

Does building any package work?  Not sure why explicitly stdplus would fail. Were you building stdplus explicitly or a complete image and that's just the package that failed? Are you building in a virtual machine and did it run out of memory?

Can you try `bitbake stdplus -c cleanall` and try again?

zyt888, 2024-09-26:
> This issue is completely garbled formatting and you didn't follow the issue template...
>
> Does building any package work? Not sure why explicitly stdplus would fail. Were you building stdplus [...]
I tried bitbake stdplus -c cleanall and try again， the problem is still exist.
Here's the error message after trying BitBake again.

NOTE: Executing Tasks
ERROR: stdplus-0.1+gitAUTOINC+6047970d6e-r1 do_configure: meson failed
ERROR: stdplus-0.1+gitAUTOINC+6047970d6e-r1 do_configure: ExecutionError('/data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/temp/run.do_configure.516670', 1, None, None)
ERROR: Logfile of failure stored in: /data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/temp/log.do_configure.516670
Log data follows:
| /data/zhangl/openbmc/poky/bitbake/lib/bb/codeparser.py:258: DeprecationWarning: ast.Str is deprecated and will be removed in Python 3.14; use ast.Constant instead
|   if isinstance(node.args[0], ast.Str):
| /data/zhangl/openbmc/poky/bitbake/lib/bb/codeparser.py:259: DeprecationWarning: Attribute s is deprecated and will be removed in Python 3.14; use value instead
|   varname = node.args[0].s
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/data/zhangl/openbmc/meta-openembedded/meta-oe/recipes-support/fmt/fmt_9.1.0.bb:do_populate_sysroot', '/data/zhangl/openbmc/meta-openembedded/meta-oe/recipes-support/function2/function2_4.2.2.bb:do_populate_sysroot', '/data/zhangl/openbmc/meta-openembedded/meta-oe/recipes-support/liburing/liburing_2.3.bb:do_populate_sysroot', '/data/zhangl/openbmc/meta/recipes-core/glibc/glibc_2.37.bb:do_populate_sysroot', '/data/zhangl/openbmc/meta/recipes-devtools/gcc/gcc-cross_12.2.bb:do_populate_sysroot', '/data/zhangl/openbmc/meta/recipes-devtools/gcc/gcc-runtime_12.2.bb:do_populate_sysroot', '/data/zhangl/openbmc/meta/recipes-devtools/qemu/qemu-native_7.2.0.bb:do_populate_sysroot', '/data/zhangl/openbmc/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', 'virtual:native:/data/zhangl/openbmc/meta/recipes-devtools/meson/meson_1.0.1.bb:do_populate_sysroot', 'virtual:native:/data/zhangl/openbmc/meta/recipes-devtools/ninja/ninja_1.11.1.bb:do_populate_sysroot', 'virtual:native:/data/zhangl/openbmc/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/data/zhangl/openbmc/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', 'virtual:native:/data/zhangl/openbmc/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/data/zhangl/openbmc/meta/recipes-devtools/python/python3_3.11.2.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['fmt', 'function2', 'liburing', 'glibc', 'gcc-cross-arm', 'gcc-runtime', 'qemu-native', 'quilt-native', 'meson-native', 'ninja-native', 'patch-native', 'pkgconfig-native', 'pseudo-native', 'python3-native', 're2c-native', 'linux-libc-headers', 'xz-native', 'gmp-native', 'binutils-cross-arm', 'libmpc-native', 'mpfr-native', 'zlib-native', 'flex-native', 'gnu-config-native', 'texinfo-dummy-native', 'zstd-native', 'libtool-native', 'glib-2.0-native', 'attr-native', 'libgcc', 'sqlite3-native', 'gdbm-native', 'util-linux-libuuid-native', 'libnsl2-native', 'bzip2-native', 'libffi-native', 'libtirpc-native', 'openssl-native', 'readline-native', 'ncurses-native', 'python3-build-native', 'python3-installer-native', 'python3-wheel-native', 'python3-setuptools-native', 'gettext-minimal-native', 'm4-native', 'gettext-native', 'libpcre2-native', 'util-linux-native', 'perl-native', 'python3-pyproject-hooks-native', 'python3-packaging-native', 'python3-flit-core-native', 'libcap-ng-native', 'make-native', 'unzip-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_configure
| NOTE: Executing meson -Dexamples=false -Dtests=disabled -Dgtest=disabled -Db_lto=true ...
| The Meson build system
| Version: 1.0.1
| Source dir: /data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/git
| Build dir: /data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/build
| Build type: cross build
| Project name: stdplus
| Project version: 0.1
| 
| ../git/meson.build:1:0: ERROR: Executables created by cpp compiler arm-openbmc-linux-gnueabi-g++ -marm -mcpu=arm1176jz-s -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/recipe-sysroot are not runnable.
| 
| A full log can be found at /data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/build/meson-logs/meson-log.txt
| ERROR: meson failed
| WARNING: /data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/temp/run.do_configure.516670:173 exit 1 from 'exit 1'
| WARNING: Backtrace (BB generated script):
| 	#1: bbfatal_log, /data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/temp/run.do_configure.516670, line 173
[... cut ...]
| 	#4: main, /data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/temp/run.do_configure.516670, line 186
ERROR: Task (/data/zhangl/openbmc/meta-phosphor/recipes-extended/stdplus/stdplus_git.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 730 tasks of which 722 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /data/zhangl/openbmc/meta-phosphor/recipes-extended/stdplus/stdplus_git.bb:do_configure
Summary: There were 2 ERROR messages, returning a non-zero exit code.

mdmillerii, 2024-09-26:
Please enclose your output by placing a line with three backticks ` ``` ` above and below your output.  This causes characters to be shown literally instead of being interpreted as markdown.   You can learn more about markdown by clicking on the [Markdown is supported](https://docs.github.com/github/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax) link between the input box and the comment button.

mdmillerii, 2024-09-26:
Also more information about your build environment 

But first read the report from the build system:
 > A full log can be found at /data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/build/meson-logs/meson-log.txt

zyt888, 2024-09-26:
> 以及有关您的构建环境的更多信息
> 
> 但首先阅读 build 系统中的报告：
> 
> > A full log can be found at /data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/build/meson-logs/meson-log.txt
ok， thanks
here is my environment：
Linux distribution based on ARM architecture, kernel version is 4.19.90
python 3.12.2
```
Build started at 2024-09-26T06:14:48.691858
Main binary: /data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/recipe-sysroot-native/usr/bin/nativepython3

linker returned <Popen: returncode: 0 args: ['arm-openbmc-linux-gnueabi-g++', '-marm', '-mcp...>
linker stdout:
GNU ld (GNU Binutils) 2.40.20230119
Copyright (C) 2023 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of

Sanity testing C++ compiler: arm-openbmc-linux-gnueabi-g++ -marm -mcpu=arm1176jz-s -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/recipe-sysroot
Is cross compiler: True.
Sanity check compiler command line: arm-openbmc-linux-gnueabi-g++ -marm -mcpu=arm1176jz-s -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/recipe-sysroot sanitycheckcpp.cc -o sanitycheckcpp_cross.exe -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/git=/usr/src/debug/stdplus/0.1+gitAUTOINC+6047970d6e-r1 -fdebug-prefix-map=/data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/git=/usr/src/debug/stdplus/0.1+gitAUTOINC+6047970d6e-r1 -fmacro-prefix-map=/data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/build=/usr/src/debug/stdplus/0.1+gitAUTOINC+6047970d6e-r1 -fdebug-prefix-map=/data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/build=/usr/src/debug/stdplus/0.1+gitAUTOINC+6047970d6e-r1 -fdebug-prefix-map=/data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/recipe-sysroot= -fmacro-prefix-map=/data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/recipe-sysroot= -fdebug-prefix-map=/data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/recipe-sysroot-native= -fvisibility-inlines-hidden -D_FILE_OFFSET_BITS=64 -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -Wl,-z,relro,-z,now
Sanity check compile stdout:

-----
Sanity check compile stderr:

-----
Running test binary command:  /data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/meson-qemuwrapper /data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/build/meson-private/sanitycheckcpp_cross.exe

../git/meson.build:1:0: ERROR: Executables created by cpp compiler arm-openbmc-linux-gnueabi-g++ -marm -mcpu=arm1176jz-s -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/data/zhangl/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/stdplus/0.1+gitAUTOINC+6047970d6e-r1/recipe-sysroot are not runnable.

```

williamspatrick, 2024-09-26:
Do other packages work?  Just stdplus is broken?  Like, does entity-manager or bmcweb work?

zyt888, 2024-09-27:
> Do other packages work? Just stdplus is broken? Like, does entity-manager or bmcweb work?

All of these packages cannot be bitbake successfully, and all have meson build errors.

mdmillerii, 2024-09-27:
It appears that white trying to invoke the first compile stage it says you are getting another compiler invoked.   Maybe you have something like a broken ccache or shell or something that's causing the wrong binary to run.

williamspatrick, 2024-11-02:
There is nothing further we are going to do with this issue.  You haven't provided sufficient information on your environment and you're trying to build an old release.  If this is something you can recreate on a newer Linux distribution with a newer OpenBMC version, please follow up with a new task using the issue template.

closed 2024-11-02 by williamspatrick