openbmc/openbmc #4018: why OpenBMC can't use std::format

583654109, 2024-10-11:
[image]
openbmc is cpp20
i try to use std::format
but error
i'm confuse....

williamspatrick, 2024-10-11:
We use std::format. 

https://github.com/search?q=org%3Aopenbmc+%22include+%3Cformat%3E%22&type=code

You didn't use the issue template so I have no idea what revision you're trying to build against. I assume something old with a compiler that didn't yet support std::format?

583654109, 2024-10-16:
> We use std::format.
> 
> https://github.com/search?q=org%3Aopenbmc+%22include+%3Cformat%3E%22&type=code
> 
> You didn't use the issue template so I have no idea what revision you're trying to build against. I assume something old with a compiler that didn't yet support std::format?

hello
my openbmc version is:
Build Configuration:
BB_VERSION           = "2.2.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "corsica-s10b316"
DISTRO               = "openbmc-phosphor"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "arm armv7a vfp vfpv4d16 callconvention-hard"
TARGET_FPU           = "hard"

Does this version of OpenBmc still not support format functionality?

williamspatrick, 2024-10-16:
`BB_VERSION = "2.2.0"`

This isn't any version information from OpenBMC.  Maybe `git describe`?

Anyhow, based on this being the bitbake version I'm going to guess you were using the langdale branch: https://wiki.yoctoproject.org/wiki/Releases

Langdale branch used GCC 12.2: https://github.com/openbmc/openbmc/tree/langdale/poky/meta/recipes-devtools/gcc

`std::format` is considered "Text Formatting" and wasn't added until GCC 13.  https://en.cppreference.com/w/cpp/compiler_support/20

Please update to a later version of OpenBMC if you want to use `std::format`.  We are constrained by the compiler choice in the underlying Yocto release.