openbmc/openbmc #4019: No chassis details found for evb-ast2600

Vinay-Rama, 2024-10-12:
Build Configuration:
BB_VERSION           = "2.9.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-22.04"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "evb-ast2600"
DISTRO               = "openbmc-phosphor"

when request for chassis details in curl -k -H "X-Auth-Token: $token" https://${bmc}/redfish/v1/Chassis
{
  "@odata.id": "/redfish/v1/Chassis",
  "@odata.type": "#ChassisCollection.ChassisCollection",
  "Members": [],
  "Members@odata.count": 0,
  "Name": "Chassis Collection"
}
How to add a data member to chassis ? i need to link the sensor to chassis so that i can get sensor details in webui? any suggestion will be helpful.

williamspatrick, 2024-10-13:
An evb is "evaluation board". How would it have a chassis?

If you're using an evb image in some other system, you'll need to write an entity-manager config for your eeprom.

Vinay-Rama, 2024-10-13:
Thanks for your response, I m new to openbmc. But what i m trying is by default evb-ast2600 as lm75 sensor in it. In order to read the sensor value in webui. The https request will get request for redfish/v1/Chassis/<str>/Sensors. If we don't get chassis details then it won't be possible to get sensor details in webui is what i understood while tracing src code.

mdmillerii, 2024-10-28:
`Chassis` is a fairly broad category and there are several `ChassisType` that might apply.

I'd suggest one of these types might be appropriate:
   - `Card`	A loose device or circuit board intended to be installed in a system or other enclosure.
   - `Component`	A small chassis, card, or device that contains devices for a particular subsystem or function.
  
But it could also be integrated into a system or otherwise an be a more generic:
  - `Enclosure`	A generic term for a chassis that does not fit any other description.
  - `Module`	A small, typically removable, chassis or card that contains devices for a particular subsystem or function.
  - `Other`	A chassis that does not fit any of these definitions.
  - `StandAlone`	A single, free-standing system, commonly called a tower or desktop chassis.

phosphor dbus interfaces currently has a more limited subset: https://github.com/openbmc/phosphor-dbus-interfaces/blob/73c931fb942daa714bfff17e950b9d5622a25842/yaml/xyz/openbmc_project/Inventory/Item/Chassis.interface.yaml#L12

But does the FRU eeprom ship with contents?  Or would you be creating your fru data or inventory item from scratch?

Regardless, the search for chassis members in bmcweb: https://github.com/openbmc/bmcweb/blob/463a0e3e880340a119bf986bab039d1afbe1a432/redfish-core/lib/chassis.hpp#L206

williamspatrick, 2024-11-02:
This is a support request.  Not a valid issue.

closed 2024-11-02 by williamspatrick