openbmc/openbmc #4020: the image  is no /lib/modules directory

xht8015, 2024-10-29:
Hello, I have a question. I want to modify some configurations as kernel modules, such as CONFIG_SENSORS_ADM1275=m. I changed the configuration file to CONFIG_MODULES=y. After building the image, zcat /proc/config.gz shows that both options are correct, but the image does not contain the adm1275.ko file, and there is no /lib/modules directory. Is there any other configuration needed?

williamspatrick, 2024-11-02:
This is a support request.  Not a valid issue.

closed 2024-11-02 by williamspatrick

xht8015, 2024-11-04:
It means that the kernel module cannot be compiled as a KO file？

mdmillerii, 2024-11-04:
That is correct.   The kernel is custom built to requirements.   We don't support modules.  Among the reasons there's no reason to waste the memory and the flash space.  The reasons have been discussed in the past.  In addition, the code building the image is public.

williamspatrick, 2024-11-04:
Bletchley uses modules in order to use the MDIO utilities. I don't think there is anything special that needed to be done.

williamspatrick, 2024-11-04:
> It means that the kernel module cannot be compiled as a KO file？ @williamspatrick

I closed it because we do not give support help via GitHub issues. This isn't a bug but a support request. We have GitHub templates for you to fill out and you ignored them.