openbmc/openbmc #4023: How to debug 'sometimes the specific sensors get no reading after AC cycle the system' issue?

jackylee0908, 2024-12-04:
Hi sir,

We have an Intel Birch Stream platform, and our BMC FW developer are implementing OpenBMC onto it with a DC-SCM module, the BMC chip is ASPEED AST2600 and the RoT is ASPEED AST1060.

We got an issue that sometimes the specific sensors get no reading after AC power cycle the system, the failure rate is about 12%, below is the example log:

CPU1_PVCCA_EHV           | 00h | ok  |  0.1 | 0.39 Amps
CPU1_PVCCA_EHV           | 01h | ok  |  0.1 | 2 Amps
CPU1_PVCCD0              | 02h | ok  |  0.1 | 0.16 Amps
CPU1_PVCCD0              | 03h | ok  |  0.1 | 2 Amps
CPU1_PVCCFA_EHV_         | 04h | ok  |  0.1 | 3.90 Amps
CPU1_PVCCFA_EHV_         | 05h | ok  |  0.1 | 29 Amps
CPU1_PVCCINF             | 06h | ok  |  0.1 | 1.25 Amps
CPU1_PVCCINF             | 07h | ok  |  0.1 | 17 Amps
CPU1_PVNN                | 08h | ok  |  0.1 | 0.08 Amps
CPU1_PVNN                | 09h | ok  |  0.1 | 1 Amps
CPU1_VCCIN               | 0Ah | ok  |  0.1 | 255 Amps
CPU2_PVCCA_EHV           | 0Bh | ok  |  0.1 | 0.39 Amps
CPU2_PVCCA_EHV           | 0Ch | ok  |  0.1 | 2 Amps
CPU2_PVCCD0              | 0Dh | ok  |  0.1 | 0.31 Amps
CPU2_PVCCD0              | 0Eh | ok  |  0.1 | 3 Amps
CPU2_PVCCFA_EHV_         | 0Fh | ok  |  0.1 | 4.76 Amps
CPU2_PVCCFA_EHV_         | 10h | ok  |  0.1 | 26 Amps
CPU2_PVCCINF             | 11h | ok  |  0.1 | 1.09 Amps
CPU2_PVCCINF             | 12h | ok  |  0.1 | 15 Amps
CPU2_PVNN                | 13h | ok  |  0.1 | 0.08 Amps
CPU2_PVNN                | 14h | ok  |  0.1 | 1 Amps
CPU2_VCCIN               | 15h | ok  |  0.1 | 255 Amps
FAN0_INLET_PWM           | 16h | ok  |  0.1 | 29.79 unspecifi
FAN0_OUTLET_PWM          | 17h | ok  |  0.1 | 29.79 unspecifi
FAN1_INLET_PWM           | 18h | ok  |  0.1 | 29.79 unspecifi
FAN1_OUTLET_PWM          | 19h | ok  |  0.1 | 29.79 unspecifi
FAN2_INLET_PWM           | 1Ah | ok  |  0.1 | 29.79 unspecifi
FAN2_OUTLET_PWM          | 1Bh | ok  |  0.1 | 29.79 unspecifi
FAN3_INLET_PWM           | 1Ch | ok  |  0.1 | 29.79 unspecifi
FAN3_OUTLET_PWM          | 1Dh | ok  |  0.1 | 29.79 unspecifi
FAN0_INLET_TACH          | 1Eh | ok  |  0.1 | 5292 RPM
FAN0_OUTLET_TACH         | 1Fh | ok  |  0.1 | 4508 RPM
FAN1_INLET_TACH          | 20h | ok  |  0.1 | 5390 RPM
FAN1_OUTLET_TACH         | 21h | ok  |  0.1 | 4508 RPM
FAN2_INLET_TACH          | 22h | ok  |  0.1 | 5390 RPM
FAN2_OUTLET_TACH         | 23h | ok  |  0.1 | 4606 RPM
FAN3_INLET_TACH          | 24h | ok  |  0.1 | 5390 RPM
FAN3_OUTLET_TACH         | 25h | ok  |  0.1 | 4606 RPM
CPU1_PVCCA_EHV           | 26h | ok  |  0.1 | 0 Watts
CPU1_PVCCA_EHV           | 27h | ok  |  0.1 | 0 Watts
CPU1_PVCCFA_EHV_         | 28h | ok  |  0.1 | 59 Watts
CPU1_PVCCFA_EHV_         | 29h | ok  |  0.1 | 47.20 Watts
CPU1_PVCCINF             | 2Ah | ok  |  0.1 | 11.80 Watts
CPU1_PVCCINF             | 2Bh | ok  |  0.1 | 11.80 Watts
CPU1_VCCIN               | 2Ch | ok  |  0.1 | 82.60 Watts
CPU1_VCCIN               | 2Dh | ok  |  0.1 | 70.80 Watts
CPU2_PVCCA_EHV           | 2Eh | ok  |  0.1 | 0 Watts
CPU2_PVCCA_EHV           | 2Fh | ok  |  0.1 | 0 Watts
CPU2_PVCCFA_EHV_         | 30h | ok  |  0.1 | 47.20 Watts
CPU2_PVCCFA_EHV_         | 31h | ok  |  0.1 | 47.20 Watts
CPU2_PVCCINF             | 32h | ok  |  0.1 | 11.80 Watts
CPU2_PVCCINF             | 33h | ok  |  0.1 | 11.80 Watts
CPU2_VCCIN               | 34h | ok  |  0.1 | 82.60 Watts
CPU2_VCCIN               | 35h | ok  |  0.1 | 70.80 Watts
Cpu_Power_Averag         | 36h | ok  |  0.1 | 124 Watts
**Cpu_Power_Averag         | 37h | ns  |  0.1 | No Reading**
Cpu_Power_Cap_CP         | 38h | ok  |  0.1 | 0 Watts
**Cpu_Power_Cap_CP         | 39h | ns  |  0.1 | No Reading**
Dimm_Power_Avera         | 3Ah | ok  |  0.1 | 300 Watts
**Dimm_Power_Avera         | 3Bh | ns  |  0.1 | No Reading**
Dimm_Power_Cap_C         | 3Ch | ok  |  0.1 | 0 Watts
CPU1_PVCCA_Contr         | 3Eh | ok  |  0.1 | 34 degrees C
CPU1_PVCCA_EHV           | 3Fh | ok  |  0.1 | 34 degrees C
CPU1_PVCCD0              | 40h | ok  |  0.1 | 42 degrees C
[... 98 lines cut ...]
CPU2_PVNN                | A3h | ok  |  0.1 | 1 Volts
CPU2_VCCIN               | A4h | ok  |  0.1 | 2 Volts
V_DCSCMB_P1V05_U         | A5h | ok  |  0.1 | 1.05 Volts
V_DCSCMB_P1V0            | A6h | ok  |  0.1 | 1.00 Volts
V_DCSCMB_P3V3_RG         | A7h | ok  |  0.1 | 3.29 Volts
V_DCSCMB_P3V3_ST         | A8h | ok  |  0.1 | 3.29 Volts
V_DCSCMB_P12V_AU         | A9h | ok  |  0.1 | 12.20 Volts
V_HPM_P1V0_AUX           | AAh | ok  |  0.1 | 0.99 Volts
V_HPM_P1V1_AUX           | ABh | ok  |  0.1 | 1.09 Volts
V_HPM_P1V2_MAX10         | ACh | ok  |  0.1 | 1.20 Volts
V_HPM_P1V8_AUX           | ADh | ok  |  0.1 | 1.78 Volts
V_HPM_P2V5_MAX10         | AEh | ok  |  0.1 | 2.47 Volts
V_HPM_P3V3               | AFh | ok  |  0.1 | 3.27 Volts
V_HPM_P3V3_AUX           | B0h | ok  |  0.1 | 3.27 Volts
V_HPM_P5V_AUX            | B1h | ok  |  0.1 | 2.79 Volts
V_HPM_P12V               | B2h | ok  |  0.1 | 12.18 Volts
V_HPM_P12V_AUX           | B3h | ok  |  0.1 | 12.18 Volts
V_HPM_P12V_STBY          | B4h | ok  |  0.1 | 11.92 Volts
V_HPM_PVCC3V3_AU         | B5h | ok  |  0.1 | 3.27 Volts

And our EE thought that it is not a HW issue and request our BMC FW developer to debug it.

Because this issue only happened with AC cycle the system, it could not be reproduced with DC power cycling test which the BMC FW has not to reboot its firmware OS, so we think it is possible to cause by BMC firmware issue, but we don't know how to debug it thru BMC firmware even the console log, we need your help to provide some directions on debugging it, thank you.

BTW, the OS we used on the system is Rocky Linux 9.4, and the sensor list was captured from the OS thru ipmitool during the test.

williamspatrick, 2024-12-04:
This is not the right forum for support.

closed 2024-12-04 by williamspatrick

jackylee0908, 2024-12-04:
> This is not the right forum for support.

Please advise where the right forum it is, thank you.

williamspatrick, 2024-12-04:
> > This is not the right forum for support.
> 
> Please advise where the right forum it is, thank you.

When you created this issue there were issue templates that you ignored. Both of them suggest for "support" style questions to use Discord or the mailing list.

jackylee0908, 2024-12-04:
> > > This is not the right forum for support.
> > 
> > 
> > Please advise where the right forum it is, thank you.
> 
> When you created this issue there were issue templates that you ignored. Both of them suggest for "support" style questions to use Discord or the mailing list.

Understood, thanks.