openbmc/openbmc #4024: Build for dl385-g11 fails with DISTRO 'poky' not found

ismaws, 2024-12-13:
I'm trying to build for dl385-g11, but I'm getting the following error after fixing some conf.local issues:

```
ismael@quiet-lemur:~/obmc/openbmc/build/dl385-g11$ bitbake obmc-phosphor-image
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    DISTRO 'poky' not found. Please set a valid DISTRO in your local.conf

Summary: There was 1 ERROR message, returning a non-zero exit code.
```

ismaws, 2024-12-13:
ismael@quiet-lemur:~/obmc/openbmc/build/dl385-g11$ cat /home/ismael/obmc/openbmc/build/dl385-g11/conf/local.conf | grep DISTRO
DISTRO ?= "poky"

williamspatrick, 2024-12-13:
Can you explain what code revision you are running, how you set up the build environment and what changes you made to local.conf?  You shouldn't even need poky.

ismaws, 2024-12-13:
I git cloned from the latest rev, followed the dev set up env guide (https://github.com/openbmc/docs/blob/master/development/dev-environment.md) and ran 

. setup dl385-g11
 
ismael@quiet-lemur:~/obmc/openbmc/build/dl385-g11$ bitbake obmc-phosphor-image
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    DISTRO 'poky' not found. Please set a valid DISTRO in your local.conf

williamspatrick, 2024-12-13:
I don't work for HPE and this machine doesn't seem super complete or maintained.  I don't know why they were referencing "poky" but none of the systems should be doing that.  Made a trivial fix here:

remote:   https://gerrit.openbmc.org/c/openbmc/openbmc/+/76565 meta-hpe: dl385: fix poky reference [NEW]

Good chance there are either still other issues you'll run into or the image might not be 100% functional on that system.   There are really only a few commits that were ever made for that machine providing anything of substance.

```
$ git log --oneline meta-hpe/meta-dl385-g11/
5d32477dee7 (HEAD) meta-hpe: dl385: fix poky reference
91a1cec28e0 styhead: update WORKDIR usage
5b0f9289bd7 meta-hpe: remove BBLAYERS_NON_REMOVABLE
e32643cc1d2 treewide: remove meta-poky usage
a4efaf7ad34 meta-hpe: add phosphor-pid-control recipe  <---
4ef8b201980 meta-hpe: adding meta-dl385-g11 layer for HPE DL385 Gen11 <---
1ab28858848 meta-hpe: dl385-g11 power supplies monitoring <---
```

ismaws, 2024-12-14:
@williamspatrick thanks for the fix, build goes a long way but ends up failing at the end with error: "hpe-uboot-signing-key.pem: No such file or directory"

```
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing python function do_generate_static
| DEBUG: Executing shell function do_generate_hpe_image
| 256+0 records in
| 256+0 records out
| 262144 bytes (262 kB, 256 KiB) copied, 0.0013356 s, 196 MB/s
| 256+0 records in
| 256+0 records out
| 262144 bytes (262 kB, 256 KiB) copied, 0.00102095 s, 257 MB/s
| Could not open file or uri for loading private key from /home/ismael/obmc/openbmc/build/dl385-g11/tmp/deploy/images/dl385-g11/hpe-uboot-signing-key.pem: No such file or directory
| WARNING: exit code 1 from a shell command.
| DEBUG: Python function do_generate_static finished
ERROR: Task (/home/ismael/obmc/openbmc/meta-phosphor/recipes-phosphor/images/obmc-phosphor-image.bb:do_generate_static) failed with exit code '1'
NOTE: Tasks Summary: Attempted 6155 tasks of which 12 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/ismael/obmc/openbmc/meta-phosphor/recipes-phosphor/images/obmc-phosphor-image.bb:do_generate_static
Summary: There were 9 WARNING messages.
Summary: There was 1 ERROR message, returning a non-zero exit code.
```

williamspatrick, 2024-12-14:
Your best bet is probably to ask in #hpe channel on Discord. Since this isn't one of the systems that is test in CI, it doesn't have first class support.