openbmc/openbmc #4027: openBMC on raspberry Pi 4

VinayWolf, 2025-01-06:
Hello all,
I was trying to build openBMC for raspberry pi. I was following the https://gerrit.openbmc.org/c/openbmc/meta-evb/+/17058/1/meta-evb-raspberrypi/README.md#29. When i do bitbake that time i m getting an error.

WARNING: /home/vinay/openBMC/openbmc/meta/lib/oe/maketype.py:99: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated since Python 3.3, and in 3.10 it will stop working
  if not isinstance(obj, collections.Callable):

Loading cache: 100% |#########################################################################################################################################| Time: 0:00:00
Loaded 3073 entries from dependency cache.
WARNING: /home/vinay/openBMC/openbmc/meta-evb/meta-evb-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.18.bb: Exception during build_dependencies for PKG_kernel-base:--
WARNING: /home/vinay/openBMC/openbmc/meta-evb/meta-evb-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.18.bb: Error during finalise of /home/vinay/openBMC/openbmc/meta-evb/meta-evb-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.18.bb
ERROR: ExpansionError during parsing /home/vinay/openBMC/openbmc/meta-evb/meta-evb-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.18.bb                 | ETA:  0:00:00
Traceback (most recent call last):
bb.data_smart.ExpansionError: Failure expanding variable PKG_kernel-base, expression was kernel-${@legitimize_package_name('${@get_kernelversion_headers('/home/vinay/openBMC/openbmc/build/tmp/work/raspberrypi_cm3-openbmc-linux-gnueabi/linux-raspberrypi/1_4.18+git${SRCPV}-r0/linux-raspberrypi_cm3-standard-build')}')} which triggered exception SyntaxError: invalid syntax (PKG_kernel-base, line 1)

Summary: There were 3 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

So what can be the possible issue/solution??
i can see that this guy tried n it worked.
https://github.com/openbmc/openbmc/issues/3581#issuecomment-513991096

mdmillerii, 2025-01-06:
This looks to be an issue with meta -raspberrypi and not the integration with phosphor and the rest of the OpenBMC project.

You can test this hypothesis by building an image according to documentation for an upstream build.  After confirming bring it to with them.

The OpenBMC project synchronizes with upstream periodically and will pick up changes made there, although you may point out recent changes that are needed. We occasionally backport individual changesets as patches,(but generally prefers to just sync again mus release.

iankao0000, 2025-02-20:
Hello everyone,
I had a problem with trying to build openbmc on raspberrypi 4 too ,
This conf 
        TEMPLATECONF=meta-evb/meta-evb-raspberrypi/conf/templates/default  . openbmc-env
And stuck at 
WARNING: linux-raspberrypi-1_6.6.63+git-r0 do_fetch: Failed to fetch URL git://github.com/raspberrypi/linux.git;name=machine;branch=rpi-6.6.y;protocol=https, attempting MIRRORS if available
ERROR: Fetcher failure: Fetch command 
export PSEUDO_DISABLED=1; 
export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus";
export SSH_AUTH_SOCK="/run/user/1000/keyring/ssh";
export PATH="/home/ian/openbmc/scripts:/home/ian/openbmc/build/tmp/work/raspberrypi4-openbmc-linux-gnueabi/linux-raspberrypi/6.6.63+git/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi:/home/ian/openbmc/build/tmp/work/raspberrypi4-openbmc-linux-gnueabi/linux-raspberrypi/6.6.63+git/recipe-sysroot/usr/bin/crossscripts:/home/ian/openbmc/build/tmp/work/raspberrypi4-openbmc-linux-gnueabi/linux-raspberrypi/6.6.63+git/recipe-sysroot-native/usr/sbin:/home/ian/openbmc/build/tmp/work/raspberrypi4-openbmc-linux-gnueabi/linux-raspberrypi/6.6.63+git/recipe-sysroot-native/usr/bin:/home/ian/openbmc/build/tmp/work/raspberrypi4-openbmc-linux-gnueabi/linux-raspberrypi/6.6.63+git/recipe-sysroot-native/sbin:/home/ian/openbmc/build/tmp/work/raspberrypi4-openbmc-linux-gnueabi/linux-raspberrypi/6.6.63+git/recipe-sysroot-native/bin:/home/ian/openbmc/poky/bitbake/bin:/home/ian/openbmc/build/tmp/hosttools"; 
export HOME="/home/ian"; LANG=C git -c gc.autoDetach=false -c core.pager=cat -c safe.bareRepository=all -c clone.defaultRemoteName=origin clone --bare --mirror https://github.com/raspberrypi/linux.git /home/ian/openbmc/build/downloads/git2/github.com.raspberrypi.linux.git --progress failed with exit code 128, see logfile for output 

The logfile
--2025-02-19 16:47:47--  http://downloads.yoctoproject.org/mirror/sources/git2_github.com.raspberrypi.linux.git.tar.gz
Resolving downloads.yoctoproject.org (downloads.yoctoproject.org)... 37.27.135.249, 2a01:4f9:3070:3985::2
Connecting to downloads.yoctoproject.org (downloads.yoctoproject.org)|37.27.135.249|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2025-02-19 16:47:47 ERROR 404: Not Found.
--2025-02-19 16:47:48--  https://mirrors.kernel.org/yocto-sources/git2_github.com.raspberrypi.linux.git.tar.gz
Resolving mirrors.kernel.org (mirrors.kernel.org)... 139.178.88.99, 2604:1380:45e3:2400::1
Connecting to mirrors.kernel.org (mirrors.kernel.org)|139.178.88.99|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2025-02-19 16:47:49 ERROR 404: Not Found.

Asking for help.

mdmillerii, 2025-02-20:
Did you check the log file for the GitHub download that returned 128?

The meta raspberry pi layer is straight from upstream and the recipe is likely from them.   Unless there is a commit changing the recipe we have not yet merged, you need to get it resolved there.

If there's a commit changing the recipe that resolves this let us know and we can consider a subtree update.

iankao0000, 2025-02-20:
Got it, thanks for the respond.

USER0FISH, 2025-03-24:
I also encountered a problem when building PI4. It was displayed as a problem with the size of u-boot. Is there any solution?

[image]

mdmillerii, 2025-03-24:
If you don't specify a supported flash size it defaults to a relatively small single NOR flash which is much smaller than a SD card or modern USB dongle built with NAND flash technology.   Look for the ladder of image sizes for the recognized values.  There's one arbitrarly size for emmc that mostly gets out of the way (unless you are building for booting from directly attached NOR).

Also, the partition sizes are based on squeezing into the selected flash size and write chosen based on the needs of another vendor board on a dated uboot version.

This is a support question not a bug report.   Please join our discord or mailing list to continue this discussion.   https://github.com/openbmc/openbmc?tab=readme-ov-file#questions and the rest of the README.