openbmc/openbmc #4030: Error: Unable to establish IPMI v2 / RMCP+ session

E1-80656-Shreya, 2025-02-04:
I have done following test setup 
QEMU romulus
PC/Laptop Ubuntu 22.04

I am trying to run command in my host machine **ipmitool -C 17 -H 10.0.2.16 -I lanplus -U root -P 0penBmc bmc info** but i am facing error 
Error: Unable to establish IPMI v2 / RMCP+ session

My romulus machine logs are as follows
root@romulus:/usr/bin# systemctl status phosphor-ipmi-host.service
* phosphor-ipmi-host.service - Phosphor Inband IPMI
     Loaded: loaded (/usr/lib/systemd/system/phosphor-ipmi-host.service; enabled; preset: enabled)
    Drop-In: /usr/lib/systemd/system/phosphor-ipmi-host.service.d
             `-10-override.conf
     Active: active (running) since Tue 2025-02-04 06:21:29 UTC; 11min ago
 Invocation: e634d0b590c94ff399c0696fcd7e730c
   Main PID: 481 (ipmid)
        CPU: 1.674s
     CGroup: /system.slice/phosphor-ipmi-host.service
             `-481 ipmid

Feb 04 06:21:25 romulus systemd[1]: phosphor-ipmi-host.service: Failed with result ...ump'.
Feb 04 06:21:25 romulus systemd[1]: phosphor-ipmi-host.service: Consumed 2.069s CPU time.
Feb 04 06:21:26 romulus systemd[1]: phosphor-ipmi-host.service: Scheduled restart j...at 1.
Feb 04 06:21:26 romulus systemd[1]: Starting Phosphor Inband IPMI...
Feb 04 06:21:29 romulus ipmid[481]: i2c allow list file not found! file name: /usr/....json
Feb 04 06:21:29 romulus systemd[1]: Started Phosphor Inband IPMI.
Feb 04 06:21:29 romulus ipmid[481]: ERROR opening IPMI provider /usr/lib/ipmid-prov...wlist
Feb 04 06:21:30 romulus ipmid[481]: New interface mapping: xyz.openbmc_project.Ipmi...nel 1
Feb 04 06:21:33 romulus ipmid[481]: Host control timeout hit!
Feb 04 06:21:33 romulus ipmid[481]: Failed to deliver host command
Hint: Some lines were ellipsized, use -l to show in full.

williamspatrick, 2025-02-04:
You gave the systemd status for the 'host' side, but not the network side.  Are any `phosphor-ipmi-net@` services running?  Usually `@eth0` at least.

Shreya-Panwal, 2025-02-05:
For the network side i have checked the status of system services phosphor-ipmi-net@eth0.service and it is running 

root@romulus:~# systemctl status phosphor-ipmi-net@eth0.service
* phosphor-ipmi-net@eth0.service - Network IPMI daemon
     Loaded: loaded (/usr/lib/systemd/system/phosphor-ipmi-net@.service; enabled; preset: disabled)
     Active: active (running) since Wed 2025-02-05 04:34:06 UTC; 10s ago
 Invocation: 2ede7f88457a424bb4473c7c11f6255d
TriggeredBy: * phosphor-ipmi-net@eth0.socket
   Main PID: 401 (netipmid)
        CPU: 363ms
     CGroup: /system.slice/system-phosphor\x2dipmi\x2dnet.slice/phosphor-ipmi-net@eth0.service
             `-401 /usr/bin/netipmid -c eth0

Feb 05 04:34:06 romulus systemd[1]: Started Network IPMI daemon.
Feb 05 04:34:08 romulus netipmid[401]: Bind to interface: eth0

saper, 2025-02-07:
Can you share your Qemu command line? You need to forward the ports correctly to reach IPMI? 

I have just built current master, followed the excellent advice how to start qemu from https://github.com/openbmc/docs/blob/master/development/dev-environment.md#download-and-start-qemu-session and it worked right away.

lxwinspur, 2025-02-07:
I tested with Qemu (romulus), it works fine:
```
./qemu-system-arm -m 1024 -M romulus-bmc -nographic -drive file=./obmc-phosphor-image.static.mtd,format=raw,if=mtd -net nic -net user,hostfwd=:127.0.0.1:2222-:22,hostfwd=:127.0.0.1:2443-:443,hostfwd=:127.0.0.1:2200-:2200,hostfwd=udp:127.0.0.1:2623-:623,hostname=qemu
```

```
romulus$ ipmitool -H 127.0.0.1 -p 2623  -C 17 -U root -P 0penBmc -I lanplus  mc info
Device ID                 : 0
Device Revision           : 0
Firmware Revision         : 2.17
IPMI Version              : 2.0
Manufacturer ID           : 0
Manufacturer Name         : Unknown
Product ID                : 0 (0x0000)
Product Name              : Unknown (0x0)
Device Available          : yes
Provides Device SDRs      : yes
Additional Device Support :
    Sensor Device
    SEL Device
    FRU Inventory Device
    Chassis Device
Aux Firmware Rev Info     : 
    0x00
    0x00
    0x00
    0x00
```

Shreya-Panwal, 2025-02-07:
Thanks all for your help. Now I am able to see the proper output of the command.