openbmc/openbmc #4034: evt-ast2500: mapper wait is stuck waiting for /org/openbmc/control/power0

saper, 2025-02-13:
**Unexpected behavior you saw**

The system is not fully up, mapper-wait@-org-openbmc-control-power0.service is waiting for `/org/openbmc/control/power0` to appear on dbus.

```
root@evb-ast2500:~# systemctl list-jobs
JOB  UNIT                                                TYPE  STATE  
119  phosphor-reset-chassis-running@0.service            start waiting
121  obmc-power-reset-on@0.target                        start waiting
151  obmc-fru-fault-monitor.service                      start waiting
118  obmc-chassis-powerreset@0.target                    start waiting
152  mapper-wait@-xyz-openbmc_project-led-groups.service start running
1    multi-user.target                                   start waiting
124  phosphor-reset-chassis-on@0.service                 start waiting
95   phosphor-discover-system-state@0.service            start waiting
138  obmc-led-group-start@bmc_booted.service             start waiting
3424 xyz.openbmc_project.LED.GroupManager.service        start running
123  mapper-wait@-org-openbmc-control-power0.service     start running

11 jobs listed.
```

(the problem with `xyz.openbmc_project.LED.GroupManager.service` has been reported separately as https://github.com/openbmc/phosphor-led-manager/issues/14)

**Expected behavior**
No systemd jobs should be running after the system startup.

**To Reproduce**
Steps to reproduce the behavior:
1. Build evb-ast2500 from the master branch
2. run under qemu-system-arm
3. after the system starts, examine `systemctl list-jobs`

**Screenshots**
If applicable, add screenshots to help explain your problem.

**OpenBMC Information:**

Built from master at e135a1e3015c8446f88067339a6746c1a4e2f344 

 - /etc/os-release: 
```
ID=openbmc-phosphor
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.18.0-dev"
VERSION_ID=2.18.0-dev-455-ge135a1e301
VERSION_CODENAME="scarthgap"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.18.0-dev"
CPE_NAME="cpe:/o:openembedded:openbmc-phosphor:2.18.0-dev-455-ge135a1e301"
BUILD_ID="20250212213005"
OPENBMC_TARGET_MACHINE="evb-ast2500"
EXTENDED_VERSION="2.18.0-dev-455-ge135a1e301"
```
   
 - uname -a: `Linux evb-ast2500 6.6.75-2f18c25-00366-g2f18c25e4b56 #1 Tue Feb  4 23:53:45 UTC 2025 armv6l GNU/Linux`
 
 - Any additional patches or changes:
   1. https://gerrit.openbmc.org/c/openbmc/phosphor-host-ipmid/+/78174
   1. https://github.com/openbmc/phosphor-led-manager/commit/44c2777e26322b03e3b4feffcff006a6e91d13bc

**Additional context**

williamspatrick, 2025-02-13:
An “evb” isn’t a real server. Some of the code for managing a real server isn’t configured “correctly” (and really never can).

saper, 2025-02-13:
Yes, that is obviously true. I am new to this code base but I have noticed things like [MACHINEOVERRIDES](https://github.com/openbmc/openbmc/blob/e5ba6f8e629103290f5f242c806cad6f909cfd6e/meta-phosphor/conf/machine/include/obmc-evb-common.inc#L4) that are supposed to tell the build system that no real host is there.

I could not find the place where such services could simply be disabled in case our host is `phospor-evb` or something like that.

There are quite a few machine checks that succeed on evb without any problems, only this one and https://github.com/openbmc/phosphor-led-manager/issues/14 seem to hang.

mdmillerii, 2025-02-13:
This sounds like an opportunity for you to develop patches and [contribute](https://github.com/openbmc/docs/blob/master/CONTRIBUTING.md) this [feature](https://github.com/openbmc/openbmc/blob/e5ba6f8e629103290f5f242c806cad6f909cfd6e/.github/ISSUE_TEMPLATE/bug_report.md?plain=1#L11...L14) back to the community.

There are at least two power control providers in use by different platforms in the project (`x86-power-control` and `phosphor-state-manager`). This can give you a starting point to explore.  In addition check out the documentation for yocto.

>  in case our host is phospor-evb

Nit this is a board not a host, as you noted there isn't one in the prior paragraph.

saper, 2025-02-14:
Thank you, I am new to all this and I was only wondering why my `CurrentBMCState` is not `xyz.openbmc_project.State.BMC.BMCState.Ready` as every OpenBMC tutorial says it should be...

It took me a while to figure out that `phosphor-state-manager` is [pushing out systemd dependencies](https://github.com/openbmc/phosphor-state-manager/blob/4fae22ad0a1df18249523cf518993c395e042562/service_files/obmc-power-stop%40.service#L7). 

But I can find providers for the `/org/openbmc/control/power0` - how to locate the code needed?

Do you think that a dummy provider that simulates host power state would be a good solution? maybe already there is one?