openbmc/openbmc #4039: phosphor-ipmi-host service should have service dependency on PAM Module ?

hirenpitroda, 2025-03-26:
Hello There,

I see in phosphor-ipmi-host service has no such dependency on PAM Module while few code files under openbmc/phosphor-host-ipmid uses PAM module apis for user authentication.
There is a possible use case where, phosphor-ipmi-host service comes up early while PAM module related service comes later, in this case, while PAM Module service coming up, in between user sends request via IPMI, it possibly gives undefined behavior.
hence there must be some PAM module related service dependency added to IPMI service(s) and all other services (WEB UI, REDFISH) which internally does authentication via PAM Module APIs.

Kindly provide valuable inputs.

mdmillerii, 2025-04-01:
When you have identified a specific issue and have a specific change to propose  (and it's not a lot of code), submitting change is the fastest way to get attention See [CONTRIBUTING.md](https://github.com/openbmc/docs/blob/master/CONTRIBUTING.md) for instructions.  Usually you'll need to be added to your employers corporate contribtors license agreement (CLA) always follow their guidelines and procedures. And security reports have their own process.

Starting with a discussion is better when you don't know what approach to take when solving an issue.  Note that, as a project, we limit using GitHub issues for reporting specific bugs as opposed to questions or discussions.

> I see in phosphor-ipmi-host service has no such dependency on PAM Module while few code files under openbmc/phosphor-host-ipmid uses PAM module apis for user authentication. 

Can you be more specific?   I purposely don't know much about IPMI but what I have learned includes phosphor ipmi host implements the code common to the network and the host channel , and that the host channel doesn't usually involve accounts.  PAM deals with authenticating a user, usually by confirming a username and password along with other criteria.  Also IPMI RCMP+ requires the plain text so I don't expect pam to be used, and the there are only 16 account slots so user role is probably managed outside pwm. 

What command fails?   Is it required for the host to boot and start the operating system,?   Sone users want to start the host as quickly as possible even if the BMC is not fully working.

Also just telling systemd if a dependency only requires it tries to start it , by defau in the same transaction (at the same time).   If it crashes that is ignored.  So failures need to be handled cleanly.  Usually by a l of g message with details of what failed and an error return for the IPMI command.

> There is a possible use case where, phosphor-ipmi-host service comes up early while PAM module related service comes later, in this case, while PAM Module service

What is the pam module service?    Please point out a line of code.

> coming up, in between user sends request via IPMI, it possibly gives undefined behavior. hence there must be some PAM module related service dependency added to IPMI service(s) and all other services (WEB UI, REDFISH) which internally does authentication via PAM Module APIs.
.

While bmcweb usebs phosphor user manager to query what authorization an account has, the authentication if the credentials is handled directly by the PAM system and doesn't use any dbus service.  Let's focus on IPMI.

I caution placing lots of requirements for services to be available vs clearly handling other services being unavailable.   What if the other service stops or crashes?    The default behavior of systemd is only to require a service that is a dependency be started it's ignored if it crashes.

vmauery, 2025-04-01:
The interaction with PAM from IPMI is not for authentication, it's only for password changes. And PAM is not a service so much as a library. So any time a password change request comes along, it should be fine.

mdmillerii, 2025-04-01:
> PAM is not a service so much as a library
> any time a password change request 

The above statements are aligned with my expectations.  

~Reading again were you expecting a build depenancy to require the library headers be included?~ You reference a PAM module service.   There is no such thing: we run under the Linux kernel not some capability system. 

> gives undefined behavior.

The community is concerned about encountering undefined behavior, but an unavailable service is only undesirable behavior as long as the error is properly detected and handled.

Please use other channels to request help and gain understanding of our architecture, designs, and how they integrate. You can find references to them in the [README](https://github.com/openbmc/openbmc?tab=readme-ov-file#openbmc)

closed 2025-04-02 by williamspatrick