openbmc/openbmc #4040: BUILD AND SERVE

ashika-jayaraj, 2025-03-27:
Please let me know why am I facing issue while installing node modules its showing most packages and deprecated also reached EOL. I am not able to run application its showing error

> webui-vue@0.1.0 serve
> check-node-version --package && vue-cli-service serve --mode development

 INFO  Starting development server...
 ERROR  TypeError: Cannot set properties of undefined (setting 'splitChunks')
TypeError: Cannot set properties of undefined (setting 'splitChunks')
    at configureWebpack (C:\Users\Ashika\Downloads\webui-vue-master\webui-vue-master\vue.config.js:61:37)
    at C:\Users\Ashika\Downloads\webui-vue-master\webui-vue-master\node_modules\@vue\cli-service\lib\Service.js:240:21
    at Array.forEach (<anonymous>)
    at Service.resolveWebpackConfig (C:\Users\Ashika\Downloads\webui-vue-master\webui-vue-master\node_modules\@vue\cli-service\lib\Service.js:237:30)
    at PluginAPI.resolveWebpackConfig (C:\Users\Ashika\Downloads\webui-vue-master\webui-vue-master\node_modules\@vue\cli-service\lib\PluginAPI.js:115:25)
    at serve (C:\Users\Ashika\Downloads\webui-vue-master\webui-vue-master\node_modules\@vue\cli-service\lib\commands\serve.js:48:31)
    at Service.run (C:\Users\Ashika\Downloads\webui-vue-master\webui-vue-master\node_modules\@vue\cli-service\lib\Service.js:219:12)
    at Object.<anonymous> (C:\Users\Ashika\Downloads\webui-vue-master\webui-vue-master\node_modules\@vue\cli-service\bin\vue-cli-service.js:36:9)
    at Module._compile (node:internal/modules/cjs/loader:1529:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1613:10)
    at Module.load (node:internal/modules/cjs/loader:1275:32)
    at Module._load (node:internal/modules/cjs/loader:1096:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:164:12)
    at node:internal/main/run_main_module:28:49

ashika-jayaraj, 2025-03-27:
can you send me the link of that repo? I was trying to deploy it with making necesarry changes I install vue 3 I found some issues with compatibility.

williamspatrick, 2025-04-02:
> can you send me the link of that repo? I was trying to deploy it with making necesarry changes I install vue 3 I found some issues with compatibility.

openbmc/webui-vue

closed 2025-04-02 by williamspatrick

ashika-jayaraj, 2025-04-04:
Can you let me know which version of node I should install?

ashika-jayaraj, 2025-04-05:
PS D:\OpenBMC_Vue\clone\webui-vue> npx --trace-deprecation vue-cli-service serve --mode development
 INFO  Starting development server...
 ERROR  TypeError: Cannot read properties of undefined (reading 'upgrade')
    at Array.forEach (<anonymous>)
    at Server.initialize (D:\OpenBMC_Vue\clone\webui-vue\node_modules\webpack-dev-server\lib\Server.js:1932:29)
    at Server.start (D:\OpenBMC_Vue\clone\webui-vue\node_modules\webpack-dev-server\lib\Server.js:3358:16)
PS D:\OpenBMC_Vue\clone\webui-vue>

This is the error am facing file running npm run serve or npx --trace-deprecation vue-cli-service serve --mode development
can you address the issue?