openbmc/openbmc #4041: Rootfs size exceeds maximum limit while building OpenBMC for Intel S2600GZ

tcsk12, 2025-04-03:
Title of the Issue
Error: Rootfs size exceeds maximum limit while building OpenBMC for Intel S2600GZ

Problem Description
While attempting to build OpenBMC for an Intel S2600GZ system, I encountered errors related to exceeding the maximum rootfs size during the build process. The build fails when attempting to execute the do_image_squashfs_xz and do_image_cpio tasks. Despite rebuilding, the errors persist, and I’m not sure how to proceed. I am hoping for some assistance in resolving the issue.

Error Statement
ERROR: obmc-phosphor-image-1.0-r0 do_image_squashfs_xz: The rootfs size 112664(K) exceeds IMAGE_ROOTFS MAXSIZE: 32768(K)

ERROR: obmc-phosphor-image-1.0-r0 do_image_cpio: The rootfs size 112664(K) exceeds IMAGE_ROOTFS MAXSIZE: 32768(K)

Warnings:
WARNING: obmc-phosphor-image-1.0-r8 do_rootfs: Group render has never been defined

WARNING: systemd-1_257.1-ro do_install: Using /home/root as root user's home directory is not fully supported by systemd

WARNING: systemd-1_257.1-re do_package: systemd: NOT adding alternative provide /usr/sbin/init: ${rootlibexecdir}/systemd/systemd does not exist

Additional Error after Rebuild:
ERROR: obmc-phosphor-image-1.0-r0 do_generate_static: Image '/home/olympus-nuvoton/openbmc/build/olympus-nuvoton/tmp/work/olympus_nuvoton-openbmc-linux-gnueabi/obmc-phosphor-image/1.0/deploy-obmc-phosphor-image-image-complete/obmc-phosphor-image-olympus-nuvoton.squashfs-xz' is 1953792 bytes too large!

ERROR: Task (/home/olympus-nuvoton/openbmc/meta-phosphor/recipes-phosphor/images/obmc-phosphor-image.bb:do_generate_static) failed with exit code '1'

What I Tried
-I attempted to rebuild OpenBMC multiple times, but the errors persist.
-I’ve checked the configuration files and verified the IMAGE_ROOTFS_MAXSIZE variable, but increasing it did not resolve the issue.
-I attempted to reduce the size of the root filesystem by removing unused components from the image, but that didn’t help.
-I verified that all required dependencies were installed for the build environment.

Expected Behavior
-I expected the build to complete successfully without the errors related to rootfs size or any other issues.

System Configuration
Hardware: Intel S2600GZ motherboard
OS: Ubuntu 24 (64-bit)
Memory (RAM): 62GB
Disk Space:
Total disk space: 916GB
Used space: 65GB
Available space: 805GB

Build Environment
OpenBMC Version: Latest version
Bitbake Version: BitBake Build Tool Core version 2.9.1

System Information:
RAM:
Total: 62GB
Used: 5.8GB
Free: 44GB
Available: 56GB
Disk Space:
Filesystem      Size  Used  Avail Use% Mounted on
tmpfs           6.3G  2.7M  6.3G   1% /run
/dev/sdb1       916G   65G  805G   8% /
tmpfs           32G  4.0K   32G   1% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
/dev/sda2       219G   18G  191G   9% /media/Intel_Disk
tmpfs           6.3G  2.6M  6.3G   1% /run/user/1000

Steps to Reproduce
-Clone the OpenBMC repository for Intel S2600GZ.
-Run the bitbake obmc-phosphor-image command.
-The build fails during the do_image_squashfs_xz and do_image_cpio tasks, resulting in the rootfs size error.

Request for Assistance
I kindly request your help in identifying the root cause of these errors. Any guidance on how to fix the rootfs size issue, the systemd warnings, or the missing render group would be greatly appreciated. I am looking for any insights or solutions that might help resolve this build failure.

Thank you for your time and support!

Additional Context
The build environment is using Ubuntu 24 with all necessary dependencies installed. Despite rebuilding and adjusting various configurations, I am unable to resolve these issues. I am running this build on the Intel S2600GZ platform with a custom configuration, which may involve platform-specific settings.

tcsk12, 2025-04-04:
Thank you for your response, and I’m sorry for the confusion!

To clarify, my build system is an Intel S2600GZ server, and the target system is "olympus-nuvoton". I am building the OpenBMC image to flash it onto QEMU virtual machine. The goal is to use QEMU for OpenBMC testing.

Also, I am new to OpenBMC, so I apologize if my previous problem description was unclear. I’ve modified it to provide better context.

I hope this clears up any confusion. Please let me know if you need further details or if there’s anything else I can do to help resolve the issue!

Thank you for your time and support!

ShubhabrataB, 2025-04-10:
I am getting a similar error : 

**ERROR:** obmc-phosphor-image-1.0-r0 do_generate_static:
build/olympus-nuvoton/tmp/deploy/images/olympus-nuvoton/fitImage-obmc-phosphor-initramfs-olympus-nuvoton-olympus-nuvoton' is 2734848 bytes too large!

in my conf file I have added few tunables to strip off the size of the rootfs image and this is how my conf file is loooking now 
local.conf:
IMAGE_FSTYPES = "squashfs"
SQUASHFS_COMP = "xz"
SQUASHFS_COMPRESSION_LEVEL = "9"
INHIBIT_PACKAGE_DEBUG_SPLIT = "1"
INHIBIT_PACKAGE_STRIP = "1"
EXTRA_IMAGE_FEATURES:remove = "debug-tweaks"
PACKAGECONFIG:remove = "nls"
IMAGE_FEATURES:remove = "read-only-rootfs"
IMAGE_INSTALL:remove = "packagegroup-core-full-cmdline"

# Remove docs, man pages, etc.
IMAGE_INSTALL:append = " packagegroup-core-boot"
PACKAGE_REMOVE = "man man-db docs"

bblayers.conf:

LCONF_VERSION = "8"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /media/code4fun/d6fd257c-7745-4324-a2df-089a5c13a45b/home/shubh/work/openbmc/meta \
  /media/code4fun/d6fd257c-7745-4324-a2df-089a5c13a45b/home/shubh/work/openbmc/meta-openembedded/meta-oe \
  /media/code4fun/d6fd257c-7745-4324-a2df-089a5c13a45b/home/shubh/work/openbmc/meta-openembedded/meta-networking \
  /media/code4fun/d6fd257c-7745-4324-a2df-089a5c13a45b/home/shubh/work/openbmc/meta-openembedded/meta-python \
  /media/code4fun/d6fd257c-7745-4324-a2df-089a5c13a45b/home/shubh/work/openbmc/meta-phosphor \
  /media/code4fun/d6fd257c-7745-4324-a2df-089a5c13a45b/home/shubh/work/openbmc/meta-nuvoton \
  /media/code4fun/d6fd257c-7745-4324-a2df-089a5c13a45b/home/shubh/work/openbmc/meta-google \
  /media/code4fun/d6fd257c-7745-4324-a2df-089a5c13a45b/home/shubh/work/openbmc/meta-quanta \
  /media/code4fun/d6fd257c-7745-4324-a2df-089a5c13a45b/home/shubh/work/openbmc/meta-quanta/meta-olympus-nuvoton \

Please assist if any there are any suggestions.

Thanks,
Shubh,

ShubhabrataB, 2025-04-13:
Any comments or suggestions please !!