openbmc/openbmc #4044: An error in compiling obmc-phosphor-image

a952760068, 2025-04-07:
Error in obmc-phosphor-initramfs package when compiling the master version of openbmc and executing bitbake obmc-phosphor-image. The error is "the post install intercept hook' update _ mandb' failed" when executing run_intercepts in do_rootfs phase during obmc-phosphor-initramfs. The specific error report is "CP: cannotstat'/home/BMC/work/openbmc/build/tmp/work/armv7ahf-vfpv4d16-openbmc-Linux-gnueabi/man-db/2.11.2/image/etc/ Man _ db.conf': no such file or directory ",but obmc-phosphor-image can compile and generate image in other os. By comparison and printing, it is found that the Generate rootfs function in do_rootfs is not executed in other os. After that, obmc-phosphor-image can be compiled, and the os versions of both environments are "name =" Ubuntu "version =" 20.04.3 lts (Focal Fossa) "".Now, I don't know how to solve the compilation problem of bitbake obmc-phosphor-image.

[image]
[image]
[image]
[image]

mdmillerii, 2025-04-07:
This is very similar #4043 I'll presume the duplicate issue is a related to a problem with your browser.

The path names include unexpected characters.  There are at least two segments with 6-20 random characters at the end of a directory name.   I suspect one or more of the follow is occurring: 
1. your storage is corrupted, 
2. you are pushing your CPU beyond the environmental limits in which it properly functions, or 
3. you are trying to use some software or hardware that is broken in some other way 
 - especially since you state it works *_in other os_* 
 - likely referring to a different physical machine running different software.

williamspatrick, 2025-04-07:
Please follow GitHub issue template.