openbmc/openbmc #4045: Building fails using x11spi configuration

3mkusiak, 2025-04-07:
**Unexpected behavior you saw**
`x11spi` configuration fails to build due to squashfs-xz being too large.

Logs:
[openbmc_x11spi.log](https://github.com/user-attachments/files/19632182/openbmc_x11spi.log)
[openbmc_x11spi.txt](https://github.com/user-attachments/files/19632190/openbmc_x11spi.txt)

**Expected behavior**
`x11spi` default configuration builds successfully.

**To Reproduce**
1. Clone `openbmc` repo from GitHub.
2. Run `setup` script for `x11spi`
  ```
  source setup x11spi
  ```
3. Attempt to build OpenBMC.
```
bitbake obmc-phosphor-image
```

**OpenBMC Information:**
- Attempted to build from master branch commit: 9a56e24a84b446a58a61e824d78b7366b729e127

**Additional context**
- Tried building from from [2.18.0-dev tag](https://github.com/openbmc/openbmc/tree/2.18.0-dev) too, no luck either. Attached log is from this attempt.

macpijan, 2025-04-08:
What's the size of resulting squasfs?
Do we know what is the flash size we can use on this platform?

3mkusiak, 2025-04-08:
> What's the size of resulting squasfs?

Did not verify if it is successfully created, but it's got `520192 bytes` overhead.
> Do we know what is the flash size we can use on this platform?

I suppose 32768k.

I managed to build custom image, based on `x11spi` by creating `[...]/recipes-phosphor/images/obmc-phosphor-image.bbappend` and removing some features. Still, might be worth to make stock `x11spi` be build-able by default.