openbmc/openbmc #4046: Error compiling romulus obmc-phosphor-image following Development Environment Tutorial

jseely, 2025-04-09:
**Unexpected behavior you saw**
During a clean build of obmc-phosphor-image for the romulus machine I ran into an Error from the openpower-software-manager do_package task.
```
ERROR: openpower-software-manager-1.0+git-r1 do_package: QA Issue: openpower-software-manager: Files/directories were installed but not shipped in any package:
  /usr/share/openpower-pnor-code-mgmt
  /usr/share/openpower-pnor-code-mgmt/org.open_power.Software.Host.Updater.conf
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
openpower-software-manager: 2 installed and not shipped files. [installed-vs-shipped]
ERROR: openpower-software-manager-1.0+git-r1 do_package: Fatal QA errors were found, failing task.
ERROR: Logfile of failure stored in: /build/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/openpower-software-manager/1.0+git/temp/log.do_package.10295
ERROR: Task (/build/meta-openpower/recipes-phosphor/flash/openpower-software-manager_git.bb:do_package) failed with exit code '1'
```

**Expected behavior**
I expected a clean build and the `obmc-phosphor-image-romulus.static.mtd` file to exist in the tmp/deploy/images/romulus directory.

**To Reproduce**
1. Checkout repo
2. Build Docker image for build environment

``` Dockerfile
FROM ubuntu

RUN apt update && apt install -y git gcc g++ make file wget gawk diffstat bzip2 cpio chrpath zstd lz4 bzip2 python3 locales
RUN locale-gen "en_US.UTF-8" && dpkg-reconfigure locales && localedef -i en_US -c -f UTF-8 -A /usr/share/locale/locale.alias en_US.UTF-8

USER 1000:1000
WORKDIR /build
```
3. Run docker image mounting the git directory as the /build directory `docker run -it --rm -v $(pwd):/build openbmc-build /bin/bash`
4. `. setup romulus`
5. `bitbake obmc-phosphor-image`

**OpenBMC Information:**
Host Machine is Docker in WSL2 on Windows
```
ubuntu@6dc158b3c9de:/build$ uname -a
Linux 6dc158b3c9de 5.15.167.4-microsoft-standard-WSL2 #1 SMP Tue Nov 5 00:21:55 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
```

williamspatrick, 2025-04-09:
What commit?

@anoo1 - There were some changes recently, I think, but I’m surprised we didn’t catch this in CI since Romulus is one of the CI systems.

jseely, 2025-04-10:
Commit: 6910ff6677c4f1d47474cde3e312c21d57e69050

anoo1, 2025-04-10:
Not seeing an error myself. The `org.open_power.Software.Host.Updater.conf` file has been part of the recipe for a long time: https://github.com/openbmc/openbmc/blob/a02fb0a827e316e20de5966d49c8eb5c6c01a568/meta-openpower/recipes-phosphor/flash/openpower-software-manager_git.bb#L64

Building the latest openbmc/openbmc master branch (with HEAD currently at commit a02fb0a827 mentioned above) with the following steps:

1. Clone the build-scripts repo:
```
$ git clone https://github.com/openbmc/openbmc-build-scripts.git
$ cd openbmc-build-scripts
```

2. Create an ubuntu 24.04 container:
```
$ export CONTAINER_ONLY=true
$ export target=romulus
$ export img_tag=24.04
$ ./build-setup.sh
```

3. Build:
```
$ docker run -it --userns=keep-id --pids-limit=-1 -u $USER -v $HOME:$HOME -w $HOME localhost/openbmc/ubuntu:24.04-romulus bash
$ cd openbmc/openbmc/
$ . setup romulus
$ bitbake obmc-phosphor-image
```

jseely, 2025-04-10:
Ok, yes that container worked for me. It must have been something related to the build container I had defined above.
```
FROM ubuntu

RUN apt update && apt install -y git gcc g++ make file wget gawk diffstat bzip2 cpio chrpath zstd lz4 bzip2 python3 locales
RUN locale-gen "en_US.UTF-8" && dpkg-reconfigure locales && localedef -i en_US -c -f UTF-8 -A /usr/share/locale/locale.alias en_US.UTF-8

USER 1000:1000
WORKDIR /build
```

closed 2025-04-10 by jseely