openbmc/openbmc #4054: The TMPDIR: openbmc/openbmc/build/qemuarm/tmp can't be located on nfs.

v9035, 2025-08-05:
git pull the master. Try to build qemuarm  a basic one.

 . setup qemuarm

 bitbake obmc-phosphor-image

ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    The TMPDIR: /openbmc/openbmc/build/qemuarm/tmp can't be located on nfs.

Summary: There was 1 ERROR message, returning a non-zero exit code.

How to overcome this. trried some ways from gpt but not helping.

mdmillerii, 2025-08-06:
Bitbake requres you TMPDIR to located in a filesystem that supports inotify.  

If this doesn't help respond with what is in proc mounts for your temp.

v9035, 2025-08-06:
i am working on nfs4 disk. Local has very less storage availability.

mdmillerii, 2025-08-06:
A quick search found this advice for placing parts of the build environment on NFS:

https://lists.yoctoproject.org/g/yocto/topic/can_not_build_yocto_on_nfs/61285968

We limit issues to bugs in the code our organization maintains.  As this is a limitation with our upstream tooling and build system please close this issue.  We don't support using issues for answering general usage questions, we have other channels for that please refer to the README in this repository or the issue template for further details.

v9035, 2025-08-06:
the link you shared shows a discussion on if we can use nfs for bitbake build. If you have more you can share it will help. 
Will close this issue. Let me know what channel i can use for questions and previous discussions.

williamspatrick, 2025-08-11:
This is something we inherit from Yocto.  If you don't like that behavior you'll have to take it upstream to them.

```
poky/meta/classes-global/sanity.bbclass
319:        return "The %s: %s can't be located on nfs.\n" % (name, path)
```

closed 2025-08-11 by williamspatrick